adafruit/circuitpython #1029: mpfshell and rshell no longer work under V3.0.0 on ESP8266

StephenHeathcote, 2018-07-15:
I am a beginning user of circuit python but have some experience with micro python.  Today I flashed a new Feather Huzzah with the latest V3.0.0 firmware.  I was able to get to a repl prompt with screen without problems. I then tried to access the device with mpfshell but get an error on open

```
** Micropython File Shell v0.8.1, sw@kaltpost.de ** 
-- Running on Python 3.5 using PySerial 3.4 --

mpfs [/]> open ttyUSB0

('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nImportError: no module named \'ubinascii\'\r\n')

```
I noticed the following in the V3.0.0 release notes:

> - ubinascii is renamed binascii to match CPython. Thanks @jepler

So it seems that the failure of mpfshell is an unintended side effect of renaming ubinascii.

I cloned the mpfshell source code from github  (https://github.com/wendlers/mpfshell) and hacked it to substitute binascii for ubinascii  in four lines and got it to work

```
** Micropython File Shell v0.8.1, sw@kaltpost.de ** 
-- Running on Python 3.5 using PySerial 3.4 --

mpfs [/]> open ttyUSB0
Connected to esp8266
mpfs [/]> repl

*** Exit REPL with Ctrl+] ***>
```
So I have a work around, and perhaps with some work could make the code of mpfshell work with either "ubinascii" or "binascii"  so that it could be used with circuit python or micropython.

In the process of debugging this I tried rshell which also fails.  rshell also uses ubinascii but the code is somewhat more convoluted so my quick attempt at hacking it did not fix the problem or perhaps there are more issues in that case.   The error message did not so obviously point to ubinascii as the problem 

I know that the elimination of the "u" prefixes is  a deliberate effort to align circuit python with CPython and understand the motivation. So this is an unfortunate side effect of a planned change rather than a bug,   But equally it is not a bug in mpfshell.   

Anyway  I thought it was worth flagging for others who might encounter the same difficulty

dhalbert, 2018-07-15:
Thanks for pointing this out. We maintain the https://github.com/adafruit/ampy program, which does some of things `mpfshell` and `rshell` do. If you'd like it to provide additional features that those programs provide, let us know in an issue.

The way this is typically handled is:

Replace
```
import ubinascii
```
with
```
try:
    import ubinascii
except ImportError:
    import binascii as ubinascii
```
We actually do it the other way round in our libraries, trying to import the non-`u` form first, since that's what we expect.

I don't know if the `mpfshell` and `rshell` maintainers would take pull requests to provide compatibility with CircuitPython, but you could try.

Will close this issue after hearing from you (or not).

StephenHeathcote, 2018-07-15:
Thanks for the response. 

I will take a look at ampy. I was Just trying to use the tools I have allready learned. I’ll let you know if I have any feature requests.

I understand how to do the import to fix this in mpfshell. It’s a little trickier because mpfshell is building expressions as text strings on the host which it then passes to the target for execution. So I need to decide if it’s tidier to catch and handle the exception in the passed string or on the host. In any case if I can figure out how to do this properly I will see if the maintainer will accept the pull request.

But first I want to go play with circuit python on my new Huzzah. You guys do such cool stuff.

tannewt, 2018-10-08:
Looks like it can be handled upstream.

closed 2018-10-08 by tannewt