adafruit/circuitpython #1030: Single file download for update - Usability Enhancement

three-bits, 2018-07-15:
My usability enhancement issue for today is to reduce the effort and likelihood of mistakes by reducing the number of downloads to update a board.

As I understand, A user must -
1. Go to https://github.com/adafruit/circuitpython/releases and down a firmware build which requires two decisions, 1) type of board and 2) type of firmware.
2. Reflash the board. UF2 is really awesome.
3. Go to https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases , select the correctly file,  download the file, and prune the /lib of unnecessary files due to board constants. 
4. Copy the /lib to the root of the board fs.

This could be simplified to a single .zip download with includes the correct firmware and library combination for a given board. I need to do a bit more researching... but I believe this can be specified in the Makefile for each port.

This issues became prominent when I got a bug after updating and I experienced a repeating error when opening the serial port immediately after updating. The issue prevented me from easily using the keyboard to drop into the REPL to debug

As a new user, I spent several hours tracking down what I might have done incorrectly while selecting the necessary files and installing them for my board. Props to the file naming convention on https://github.com/adafruit/circuitpython/releases . The file name exactly matched the silkscreen on the back of my board so I least I could assume I download the correct firmware.

David

 (I believe the bug I experienced had to do with an API change between /lib and what the main.py shipped with the board expected. I'll go more into that in another bug report.)

tannewt, 2018-07-16:
Yup! This is a good idea. A website with pictures of all the boards would make it easier to filter every thing down. May not needed a single zip then.

We should also include up to date examples and demos.

dhalbert, 2021-06-15:
We have https://circuitpython.org, `circup`, project bundles, and the like now, so closing this as it is mostly done. Please open a new issue if you have further enhancement ideas.

closed 2021-06-15 by dhalbert