adafruit/circuitpython #1038: nrf: implement audioio.AudioOut

dhalbert, 2018-07-17:
The nRF52 chips don't have a DAC, so any audio output would need to be done using PWM.

tannewt, 2018-07-17:
I think we can skip this. We just can throw an error on construction.

arturo182, 2018-07-17:
I would actually like to have this at some point so maybe let's keep the issue open :)

tannewt, 2018-07-17:
Could we have a separate module for it since its not a DAC? Arduino merges the two together but I'd rather not.

dhalbert, 2018-07-17:
`pwmaudio.AudioOut` or `audioio.PWMAudioOut`? Maybe not the latter because it would be dead code on boards with DAC's? But maybe not if you want multiple voices and don't have enough DACs.

tannewt, 2018-07-23:
I'd prefer the separate module for separate availability.

tannewt, 2018-10-09:
Made this long term since its a new API.

jepler, 2019-07-15:
I have begun work on this in a branch in my personal repository, but nothing's ready for integration yet.

nrf's pwm module has some limitations:
 * 16MHz max count rate (A top value of 256 (255?) gives 62.5kHz)
 * Each PWM value can be repeated an integer number of full PWM cycles
 * When AudioOut device is constructed, will claim an entire PWM instance, like a variable-frequency PWM does (normally, can support 4 pins)

There are two general approaches to adapting an arbitrary sample rate to these limitations:
 * re-sample to 62.5kHz
 * pick a carrier rate that is an integer multiple of sample rate and above some minimum frequency, and re-sample max value accordingly

Compared to what has been discussed here, my module currently goes by `audioio`.  I'll put renaming it to `pwmaudio` on the list.  I wonder whether this means RawSample and WaveFile have to be duplicated too, or can they be shared?

tannewt, 2019-07-15:
They should be shared. Is the trouble with RawSample and WaveFile being in audioio that AudioOut is there as well and can't be implemented?

jepler, 2019-07-15:
@tannewt I think I understand how to share the implementation, it's just a detail I didn't anticipate in estimating because I hadn't read the previous discussion on this issue. My mistake, and not a big deal.

jepler, 2019-07-20:
Please see #2000

tannewt, 2019-08-23:
Fixed by #2000 . Thanks @jepler

closed 2019-08-23 by tannewt