adafruit/circuitpython #1039: nrf: implement busio.UART

dhalbert, 2018-07-17:


hathach, 2018-09-18:
While working on uart, @arturo182 is it ok now if I remove the macro that use the jlink cdc instead of native cdc. That option is there for you only :)

arturo182, 2018-09-18:
Which cdc do you mean?

hathach, 2018-09-18:
the jlink cdc that uses the hwuart on pca10056 CFG_HWUART_FOR_SERIAL
https://github.com/adafruit/circuitpython/blob/master/ports/nrf/supervisor/serial.c#L35

arturo182, 2018-09-18:
Yes, it's fine to remove that :)

hathach, 2018-09-18:
anyone knows this function got called, or it is called by system `board_uart`
https://github.com/adafruit/circuitpython/blob/master/ports/nrf/board_busses.c#L108

hathach, 2018-09-19:
@dhalbert the default uart for pca10056 should be P0.8 and P0.6 instead of P1.1 & P1.2. It is connected to jlink cdc allow for quick testing. Furthermore if jlink cdc is not connected, those pins are in tri-state, which is safe to use with other device. Please let me know if I missed something.
https://github.com/adafruit/circuitpython/blob/master/ports/nrf/boards/pca10056/mpconfigboard.h#L49

hathach, 2018-09-19:
Ah, I see, seem like you want to keep the pin compatible with Arduino layout :D . Sorry, didn't realize that until I check the pcb

dhalbert, 2018-09-19:
@hathach right, exactly.

hathach, 2018-09-19:
@dhalbert  @tannewt  should the `self->buffer` is freed in common_hal_busio_uart_deinit() , or we can just leave it to gc

https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/busio/UART.c#L217

dhalbert, 2018-09-19:
Leave it to gc. It is allocated by `gc_alloc()` as a regular garbage-collectable object.

hathach, 2018-09-19:
ah thanks, this new to me, I will just skip the free for nrf52 as well.

hathach, 2018-09-20:
probably the same hw issue that I encountering. The first 2 irq uart request will fail miserably without real reason. UART is not too easy after all ~.~  
https://devzone.nordicsemi.com/f/nordic-q-a/33677/nrf52840-pin-uart-rx-bug

dhalbert, 2018-09-20:
@hathach that devzone post is very confusing. I wonder if that's the PDK board.

dhalbert, 2018-09-20:
i could try it with a GPS

hathach, 2018-09-20:
You don't need extra hw to test since only the write() work now. The bug prevent me to do subsequent receiving since it failed at the begining.

here is the code
https://github.com/hathach/circuitpython/blob/nrf52_uart_io/ports/nrf/common-hal/busio/UART.c#L75
board.UART() will call the first nrfx_uart_rx() which will trigger error 4 (frame error) ---> nrfx will cancel transfer internally.

Just run u.read() : this will printf the previous error and call nrfx_uart_rx() again. 
Then call u.read() again to get the previous error of the 1st read which is 0x08 break error.
afterward no more error, call more will result in NRFX_BUSY code.

To test with u.write() you can hook up FTDI cable, it should work.

```
Adafruit CircuitPython 4.0.0-alpha-1079-g9c2530687-dirty on 2018-09-21; PCA10050
>>> import board                                                                
>>> u = board.UART()                                                            
>>> u.read()                                                                    
error = 0x00000004                                                              
rx count = 0x00000000                                                           
b''                                                                             
>>> u.read()                                                                    
error = 0x00000008                                                              
rx count = 0x00000000                                                           
b''                                                                             
>>> 
>>> u.read()                                                                    
rx count = 0x00000000                                                           
b''                                                                             
>>> u.read()                                                                    
rx count = 0x00000000                                                           
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
AssertionError: error = 0x0BAD000B   
```

dhalbert, 2018-09-20:
Are you using UART or UARTE?

dhalbert, 2018-09-20:
I asked for more info in that DevZone case.

hathach, 2018-09-20:
UART since the rest of repo use it. UARTE is just having DMA to move data into buffer, not really needed for cpy I guess. Also the DMA demands data must be on the SRAM, flash address (i.e global const [] ) won't work. I am not sure if cpy use any of those.

dhalbert, 2018-09-20:
What exactly is the error you are getting? Is this relevant? https://devzone.nordicsemi.com/f/nordic-q-a/37521/nrf52840-usb_cdc_acm-uart/145013#145013

I saw one DevZone case that recommended using UARTE over UART in nrf52832, but now I can't find it again.

hathach, 2018-09-20:
error 4 (frame ) then error 8 (break). oh way, I think I found my own error lol, let's me test a bit more.

hathach, 2018-09-20:
facepalm, I just assign the txd 2 time :( . That is high price to pay here ~.~ . Working late is not good at all lol
https://github.com/hathach/circuitpython/blob/nrf52_uart_io/ports/nrf/common-hal/busio/UART.c#L108

dhalbert, 2018-09-20:
no problem, but time to sleep!

hathach, 2018-09-20:
Do you prefer UARTE to UART, I could upgrade it to UARTE. Though, you must make sure all data is in SRAM not flash, and we must malloc buffer if needed. I think it is not really gain much benefit, since the uart is slow, cpu time for moving data is not much at all.

dhalbert, 2018-09-20:
No, I don't prefer, just thought that UARTE might not have the problem, and now it's moot.

hathach, 2018-09-20:
uart io should be done, I will hook an GPS wing to test tomorrow. Phew, too much time for the easy uart. Hopefully there is no more bugs :D

hathach, 2018-09-21:
No problems, I will make the switch. the flow is similar to what I am doing anyway. IRQ vs DMA only has diff that cpu must move the data manually. But I think I will test what I have first before switching. I will also update the serial.c as well. currently serial.c use polling method, so there should be no impact at all.

tannewt, 2018-09-21:
DMA can be a bit faster even when we poll because it runs even when the cpu
is handling an interrupt.

hathach, 2018-09-21:
@tannewt yeah, I am moving it to uarte now.

tannewt, 2018-09-21:
Ok great!

dhalbert, 2018-10-05:
Still to implement: using both instances on nRF52840. Add `uart_init()` to disable on soft restart. @hathach I will take these up later.

tannewt, 2019-01-11:
This has been done right?

ladyada, 2019-01-11:
its been solid so far, @jerryneedell has done the most testing

jerryneedell, 2019-01-11:
It’s been working great on the particle boards

dhalbert, 2019-01-11:
It only provides one UARTE, and there are two available. Not sure the `uart_init()` has been fixed either. See https://github.com/adafruit/circuitpython/issues/1039#issuecomment-427496583 above.

dhalbert, 2019-01-11:
Nope, no `uart_init()` yet.

jerryneedell, 2019-01-11:
oops - sorry about the premature close...

tannewt, 2019-02-14:
@dhalbert Can we close this now after #1541 ?

dhalbert, 2019-02-14:
Yes, I thought it was going to get closed automatically.

closed 2019-02-14 by dhalbert

tannewt, 2019-02-14:
It's not super smart about how when it knows to close: https://help.github.com/articles/closing-issues-using-keywords/