adafruit/circuitpython #1042: nrf: implement nvm.ByteArray

dhalbert, 2018-07-17:
Like the SAMD chips, there is no true EEPROM.

nickzoic, 2019-01-28:
So would this be built over the NVMC peripheral to write to flash?

dhalbert, 2019-01-28:
Yes, see `ports/atmel-samd/common-hal/nvm/ByteArray.*` as a model to use for an analogous functionality on nrf.

nickzoic, 2019-02-02:
Branch-in-progress: https://github.com/nickzoic/micropython/tree/circuitpython-nickzoic-1042-nrf-nvm-bytearray

tannewt, 2019-02-05:
We're only targeting the nRF52840 which should simplify the first two. The intention of `CIRCUITPY_INTERNAL_NVM_SIZE` is to match `FLASH_PAGE_SIZE`. On the SAMD21 it's 256 and on SAMD51 it's 8192.

Does that simplify things?

nickzoic, 2019-02-05:
Oh okay so '832 has a 256 byte page size? I didn't try it on that
platform yet.  It certainly removes a potential nasty situation if
CIRCUITPY_INTERNAL_NVM_SIZE is a multiple of FLASH_PAGE_SIZE.

tannewt, 2019-02-06:
SAMD21 is 256. I don't remember what the '832 is. Yup, they are related to simplify things.

nickzoic, 2019-02-10:
OK so I should be wrapping all the NVM stuff in the Makefile and elsewhere in `#if NRF52840` then?

tannewt, 2019-02-11:
@nickzoic Nope, no need because the nRF port only supports the nRF52840.

nickzoic, 2019-02-12:
@tannewt Oh okay no worries.  I'll get this finished off ASAP.

nickzoic, 2019-03-02:
OK I completely forgot about this but I think it is okay ... now i've just got to rebase it!

nickzoic, 2019-04-04:
G'day @dhalbert interested in your thoughts on the approach in the new branch above (nickzoic/micropython@625bd0 etc) ... I've split the actual page access off as per your suggestion and allocated some address space for the NVM explicitly in the `boards/*.ld` configuration ... only problem is, that doesn't really work with the `CIRCUITPY_INTERNAL_NVM_SIZE` approach.  I took the space out of the text segment instead of the FATFS segment so that an upgrade to a board with an existing FATFS wouldn't trash the filesystem ...

dhalbert, 2019-04-04:
@nickzoic I don't think it's important to preserve the existing file system, as long as we warn people on upgrade. It would only matter to PCA10059 and other boards without external flash for CIRCUITPY. It's better to be consistent with the other boards.

dhalbert, 2019-04-04:
I skimmed the refactoring you did to handle the SD-safe NVM writing. Looks good!

nickzoic, 2019-04-04:
Oh, great!  I'm not totally confident with the workings of `common.ld` so it's a bit of a wild stab ... but I thought it was clearer than having the FATFS use logic something like:
```
real_fatfs_length = __fatfs_flash_length - CIRCUITPY_INTERNAL_NVM_SIZE
nvm_origin = __fatfs_flash_origin + real_fatfs_length
```
... although that'd be okay if we want to support nvm size configurable by `CIRCUITPY_INTERNAL_NVM_SIZE`.

dhalbert, 2019-07-31:
Implemented by #1042.

closed 2019-07-31 by dhalbert

kevinjwalters, 2020-10-08:
When is this nvm-related test https://github.com/adafruit/circuitpython/blob/main/tests/circuitpython/nvm_present.py#L2-L3 executed? It has a note in it about removing a `skip_if.always()` which _might be important_?