adafruit/circuitpython #1046: nrf: implement rtc.RTC

dhalbert, 2018-07-17:
There is an RTC peripheral that runs off the low-frequency clock.

nickzoic, 2019-02-02:
Branch-in-progress: https://github.com/nickzoic/micropython/tree/circuitpython-nickzoic-1046-nrf-rtc
(very little progress yet)

nickzoic, 2019-02-07:
(weird, the `void MP_WEAK common_hal_rtc_get_time` declaration seems to be stuffing me up in the nRF port but not the atmel-samd port despite using the same gcc for both ... I suspect something to do with the linker options, will have to investigate further)

nickzoic, 2019-02-10:
Wow, the __weak rabbit hole just keeps going deeper.  If I keep the void MP_WEAK declarations in, but change *either* of the error messages, so that they are *different*, it works correctly, eg:
```
 void MP_WEAK common_hal_rtc_set_time(timeutils_struct_time_t *tm) {
-    mp_raise_NotImplementedError(translate("RTC is not supported on this board"));
+    mp_raise_NotImplementedError(translate("RTC is not supported on this board X"));
 }
```
... and suddenly the actual common_hal_rtc_get_time starts working.  I'm going to have to dig into this one a bit further, either it's a linker bug or I'm doing something massively dumb (probably the latter).

nickzoic, 2019-02-15:
OK, this hole goes even deeper: if I compile with `-O1` it works as expected.
With `-Os` or `-O2` it gives the weird `TypeError: Tuple or struct_time argument required` error.
Same result with gcc-arm-none-eabi-8-2018-q4-major and gcc-arm-none-eabi-7-2018-q2-update

dhalbert, 2019-02-15:
I am doing a ton of refactoring of the makefiles and the config system, so if you have something else to work on, you can hold off on the rabbit hole for now and we can try again when I finish. It won't necessarily fix this problem, but it might be easier to debug and try different things.

nickzoic, 2019-02-15:
Oh cool, thanks Dan.  Will do, I'll check back in next week. 
The atmel-samd port works fine with the same shared-bindings and arm
compiler so I'm a bit flummoxed.Maybe different compiler flags & stuff.

jepler, 2019-08-01:
@nickzoic @dhalbert can this issue be closed up?  It looks like it was covered by #1736.

dhalbert, 2019-08-01:
Implemented by #1046. Thanks @jepler

closed 2019-08-01 by dhalbert