adafruit/circuitpython #1049: nrf: SPI flash and QSPI flash support

dhalbert, 2018-07-17:
As used on existing SAMD Express boards.

I don't see an explicit issue for this, but maybe I missed it.

hathach, 2018-07-17:
note to myself: The flash chip is GD25Q16CTIGR right

hathach, 2018-07-17:
Will try to remember it :D. I went through the time when Nordic sdk upgrade requires massive amount of work for migration. So I often to write the code in its generic form. The new nrfx seems to do a great job :D

hathach, 2018-09-04:
QSPI on pca10056

[image]

hathach, 2018-09-08:
doing a bit of testbench, is the speed Ok 
**clock 32 Mhz**
[image]

**clock 8 Mhz**
[image]

ladyada, 2018-09-08:
cool!

hathach, 2018-09-08:
The read speed is similar to cpx test, but the write speed is not that great. I will do a bit more tweaks to see if it could improve more. 
[image]

dhalbert, 2018-09-08:
The write code -might- need work. It seems slow on SAMD51 (writing a large file takes a long time), and maybe there's a little bug somewhere.

tannewt, 2018-09-10:
Write speed is erase limited since we don't do any fancy block allocations. I would love to do balancing that erases after a write but that only helps for a single block.

tannewt, 2018-11-19:
I added this with 97bc95183d5114ce69c9387ebd2e2aa13feb65a7

closed 2018-11-19 by tannewt

ladyada, 2018-11-19:
w00p