adafruit/circuitpython #1053: nrf52 I2C does not recognize am2320 sensor

jerryneedell, 2018-07-20:
I have tried several time to connect an am2320 temperature/humidity sensor to both a feath52832 and a pca10056 Dk -- on board bards, the sensor is never detected on the I2C bus.
The same sensor works on a feather M4 Espress or feather m0 express.

However, I did note that even though the sensor works on the feather m4 express, running an i2c.scan also fails to detect the sensor!

I looked at the code and see that there are some differences in the Atmel and nrf52 implementations that may offer a clue.

the Atmel code does attempt a retry on a failed write or read:
https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/busio/I2C.c#L193

but it does not appear that this is implemented in the nerf code.

It aslo appears that the first transaction to the am2320 sensor does indeed fail. Here is a screen shot of a "good" run of the sensor on the feather m4 express but the first attempt to read from the board files and it succeeds on the retry

[image]

On the nrf feather52832 - I captured some I2C scans with and without the am2320 connected.
When the sensor is connected, there a delay after the NAK for address 0x5c (write of 0xb8)
without the am2320
[image]

with the am2320
[image]

It is not clear why the first transaction fails. Still trying to understand that...

arturo182, 2018-07-20:
dibs!

ladyada, 2018-07-20:
ive noticed this sensor is wacky and sometimes you have to 'scan' for it twice

jerryneedell, 2018-07-21:
A few more bits of information:
Env though the am2320 functions with the feather m4 express, When I run the I2C scan on the feather M4 express, it does not detect the am2320 -- the logic analyzer trace looks exactly like the one above for the nrf52 -- with the gap after the attempt to detect the am2320 
[image]

The am2320 data sheet does say that the maximum data rate is 100Kbps!!  but the default for the M4 and the nrf52 is 400KBps  -- However, I reduced the frequency setting to 100K and saw no change in the behavior. still fails i2c scan -- still sends data in normal operation.

From the am2320 data sheet 

> I 2 C bus communication rate controlled by the host, to fast to slow. However, the maximum rate
is limited, the transfer rate on a I 2 C bus data in standard mode speeds up to 100Kb / s.

The am2320 functions on the featherm4 express at bot 400 and 100KBps, but is not detected by the i2c scan at either rate.

jerryneedell, 2018-07-21:
Note: the i2c_scan function does not use the same i2c with as normal operation - it uses a "probe" function which does not do retries.
https://github.com/adafruit/circuitpython/blob/master/shared-bindings/busio/I2C.c#L137 

This explains why the i2c_scan fails but the sensor still works on the feather m4.

jerryneedell, 2018-07-21:
hmm -- maybe this is starting to make sense:
on page 16 of the am2320 data sheet http://akizukidenshi.com/download/ds/aosong/AM2320.pdf

>◎I 2 C read and write timing decomposition
Sensors read or write, you must follow these three steps, otherwise it will not communicate or can’t read the correct data:
Step one: Wake Sensor
In order to reduce the humidity sensor errors caused by self-heating, the sensor in a non-working state, dormant, so to read the sensor must wake sensor to transmit commands to read and write, otherwise the sensor will not respond. It should be noted that, in the wake sensor, I 2 C address is sent, the sensor does not respond to ACK, but the host must send back an ACK clock verify that the ninth SCL clock signal. Wake sensor operation is following instructions:
After the host sends a start signal is applied to the starting address, wait for sometime (waiting
time of at least 800 μs, the maximum 3ms; such as host hardware I 2 C, you do not need to wait for
the hardware I 2 C will automatically wait), and then sends a stop signal.
Namely: the starting signal +0 xB8 + wait (> 800us) + stop signal timing diagram shown in
Figure 15.

so it does appear that there is a "wake up" issue with the am2320 sensor.  Perhaps this explains all of the issues we are seeing. It still states a maximum clock rate of 100Kbps and we are clearly violating that. I suppose that is something to be dealt with in the am2320 guide and setup. However, as noted, it actually seems to work at 400K bps.

dhalbert, 2018-10-03:
Hi @jerryneedell: Since this seems to be an AM2320 problem, which we've worked around in the drivers, can this be closed?

ladyada, 2018-10-03:
yah i fixed i think - but please try the latest version

jerryneedell, 2018-10-03:
The sensor seems to work fine with the latest library on the RPi and on a feather_nrf52832 and the nrf52_Dongle.

It still fails to be detected by an I2C scan, but that is an issue with the way scans are done.

jerryneedell, 2018-10-03:
Closing since the sensor is working with the current drivers.

closed 2018-10-03 by jerryneedell