adafruit/circuitpython #1055: Expanding touchio class

PTS93, 2018-07-24:
I would like to expand the capabilities of the touchio class a bit by adding some of the functionality available in freetouch but not present so far in the cpy wrapper.
Is there a best practice for approaching platform specific features in an otherwise generic class? Is there any similar case in cpy/mpy that I could have a look at maybe?
Lastly is there even an interest in that or were features like oversampling left out on purpose?

tannewt, 2018-07-24:
If the feature is generic (check nRF52) then we can expand `touchio`. If not, a samd specific module thats like `touchio` could be added.

We left it out for simplicity but can add it if there is demand.

tannewt, 2018-07-25:
Here is the one platform specific module: https://github.com/adafruit/circuitpython/tree/master/ports/atmel-samd/bindings

PTS93, 2018-07-26:
And how is this then handled? Some makefile setup? Where and how is this samd clock.h then used? Does it replace an otherwise generic clocks.h that is used by other platforms?

tannewt, 2018-07-30:
Its similar to these instructions: https://circuitpython.readthedocs.io/en/3.x/docs/common_hal.html

(Sorry for the slow reply.)

PTS93, 2018-08-30:
Oh, I totally missed your reply. Sorry.
For the project I wanted to use it I ended up using Arduino with the Freetouch library but I will see if I get to that at later point, I would find it still pretty interesting for the future.

tannewt, 2018-10-08:
@PTS93 What specifically did you want added? Are you still interested in working on it?