adafruit/circuitpython #1063: print() is slower in CircuitPython v3.0 compared to v2.3.1

hchhaya, 2018-07-26:
The performance of print() in CircuitPython v3.0 is slower than in CircuitPython v2.3.1

Test case: On a Trinket M0, here's the time to print a string of 20 chars 1000 times on a Windows 7 PC

v2.3.1		0.99 s
v3.0.0		3.78 s

The test program (string20.txt) is attached
[string20.txt](https://github.com/adafruit/circuitpython/files/2232616/string20.txt)

hchhaya, 2018-08-15:
A colleague took a closer look at this and found that the mp_hal_stdout_tx_strn_cooked() function in lib/utils/stdout_helpers.c sends the output string one character at a time. This results in many small USB transactions which ends up slowing down the prints.

Changing the mp_hal_stdout_tx_strn_cooked() function to call mp_hal_stdout_tx_strn() with the complete string improves the performance.

tannewt, 2018-08-15:
@hchhaya That behavior does sound suspect. The USB stack changed between 2.x and 3.x so I suspect its the problem. This is probably the best place to start tracing the changes from: http://elixir.tronnes.org/circuitpython/latest/source/ports/atmel-samd/usb.c#L282

cpforbes, 2019-02-08:
I just ran a test on a Trinket M0 with CircuitPython 2.3.1, 3.1.2 and 4.0.0-beta.2-23-g72f0783b6

```
(name='circuitpython', version=(2, 3, 1))
1000 20 character lines
Run time = 1.575s
Done

(name='circuitpython', version=(3, 1, 2))
1000 20 character lines
Run time = 2.175s
Done

(name='circuitpython', version=(4, 0, 0))
1000 20 character lines
Run time = 0.554s
Done

```
Test main.py:
```# Let's print some strings
import sys
import time

def a(size, count):
    str = "A"*size
    
    start = time.monotonic()
    for j in range(count):
        print(j, str)

    stop = time.monotonic()
    print(sys.implementation)
    print("{} {} character lines".format(count, size))
    print("Run time = %.3fs" % (stop - start))
    print ("Done")

a(20,1000)
```

tannewt, 2019-02-08:
Thank you @cpforbes! I'll close it. @hchhaya please let me know if you disagree and we can reopen.

closed 2019-02-08 by tannewt

ladyada, 2019-02-08:
3x faster? nice  :)