adafruit/circuitpython #1065: ussl on ESP/HUZZAH

ubiquitousthey, 2018-07-27:
I spent some time digging through why I couldn't get mqtt over TLS to work on my Feather HUZZAH.  I have come to the conclusion that micropython/extmod/modussl_axtls.c is ahead of circuitpython/extmod/modussl_axtls.c.

What would it take to update this?

sommersoft, 2018-07-27:
@ubiquitousthey,

@dhalbert is currently working on pulling upstream down. This may be included, but I'm not sure. Also, it may not immediately be available in a stable release (3.x atm).

ubiquitousthey, 2018-07-27:
FYI - I checked the 3.x branch, and it is not updated.  @dhalbert, let me know if you need someone to test this when you pull it in.  I have pretty much ready to test it.  It is currently complaining to me that I am sending kwargs that are not supported.

tannewt, 2018-10-08:
@ubiquitousthey Does the 4.0.0 alpha work for you? Its been updated from MicroPython.

olober, 2018-10-20:
I tried 4.0.0 alpha 2 and used code from docs. I got "object has no attribute 'IPPROTO_SEC'" on "sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM, socket.IPPROTO_SEC)". Am I doing something wrong or is it still work in progress?

jerryneedell, 2018-10-20:
hmm - I have been using usocket, but I just noticed that checking the installed modules I get
```

Adafruit CircuitPython 4.0.0-alpha.2-11-gb436666e8-dirty on 2018-10-20; ESP module with ESP8266
>>> 
>>> help('modules')
__main__          esp               ntptime           uio
_boot             flashbdev         os                ujson
_onewire          framebuf          port_diag         upip
_webrepl          gc                pulseio           upip_utarfile
analogio          hashlib           random            ure
array             inisetup          select            uselect
binascii          json              socket            usocket
bitbangio         lwip              socket            ussl
board             machine           storage           utimeq
btree             math              struct            uzlib
builtins          microcontroller   sys               webrepl
busio             micropython       time              webrepl_setup
collections       multiterminal     uctypes           websocket
digitalio         neopixel_write    uerrno            websocket_helper
errno             network           uheapq
Plus any modules on the filesystem
>>> 
```
why is socket in there twice?

jerryneedell, 2018-10-20:
ah -- looks like a non issue  -- socket and usocket are the same module in mpconfigport.h
```
#define MICROPY_PORT_BUILTIN_MODULES \
    { MP_OBJ_NEW_QSTR(MP_QSTR_esp), (mp_obj_t)&esp_module }, \
    { MP_OBJ_NEW_QSTR(MP_QSTR_socket), (mp_obj_t)&mp_module_lwip }, \
    { MP_OBJ_NEW_QSTR(MP_QSTR_usocket), (mp_obj_t)&mp_module_lwip }, \
  ... 
#define MICROPY_PORT_BUILTIN_MODULE_WEAK_LINKS \
    ...
    { MP_ROM_QSTR(MP_QSTR_socket), MP_ROM_PTR(&mp_module_lwip) }, \

```
socket does appear twice -- is that a problem?

ladyada, 2019-08-23:
hiya, we no longer support the esp8266 as of 4.x - please check out micropython for esp8266 support! thank you :)

closed 2019-08-23 by ladyada