adafruit/circuitpython #1067: uctypes is in the docs, but not in seemingly any build

hopkapi, 2018-07-28:
So this exists: [http://circuitpython.readthedocs.io/en/3.x/docs/library/uctypes.html](http://circuitpython.readthedocs.io/en/3.x/docs/library/uctypes.html)

Yet if I try to import it on a Feather M4 I get "ImportError: no module named 'uctypes'", which implies maybe it's not actually included, despite the docs implying it does?

hopkapi, 2018-07-30:
Huh, I can't see the word unsupported anywhere, for me anyway, it says "Though this MicroPython-based library _is available for use in CircuitPython_, its functionality may change in the future, perhaps significantly.". Think my browser might be rendering the page wrong?

dhalbert, 2018-07-30:
We should have removed it from the documentation. `struct` could serve the same purpose in many cases. Would it work for you? It takes about 2760 bytes to include, so it's too big to include in a number of our builds.