adafruit/circuitpython #1069: Update nrf 52832 bootloader instructions

dhalbert, 2018-07-29:
@hathach could you confirm this suggested change? Thanks.

From @jerryneedell in https://github.com/adafruit/circuitpython/pull/1068#issuecomment-40870499:

note that instructions for loading the bootloader to the feather52832 are out-dated on
https://github.com/adafruit/circuitpython/tree/master/ports/nrf/boards/feather52832

the command is shown as
```
$ make BOARD=feather52832 SERIAL=/dev/tty.SLAB_USBtoUART SOFTDEV_VERSION=5.0.0 boot-flash
```
but it should be
```
$ make BOARD=feather52832 SERIAL=/dev/tty.SLAB_USBtoUART SOFTDEV_VERSION=5.0.0 dfu-bootloader
```
here is how I ran it (on Linux)
```
jerryneedell@Ubuntu-Macmini:~/circuitpython_master/ports/nrf$ make BOARD=feather52832  SD=s132  SOFTDEV_VERSION=5.0.0 SERIAL=/dev/ttyUSB0 dfu-bootloader
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
nrfutil --verbose dfu serial --package boards/feather52832/bootloader/feather52_bootloader_5.0.0_s132_single.zip -p /dev/ttyUSB0 -b 115200
Upgrading target on /dev/ttyUSB0 with DFU package /home/jerryneedell/projects/adafruit_github/circuitpython_master/ports/nrf/boards/feather52832/bootloader/feather52_bootloader_5.0.0_s132_single.zip. Flow control is disabled, Dual bank mode
Starting DFU upgrade of type 3, SoftDevice size: 136360, bootloader size: 22744, application size: 0
Sending DFU start packet
Sending DFU init packet
Sending firmware file
#########################################################################################################################################################################################################################################################################################################################
Activating new firmware

DFU upgrade took 26.2996571064s
Device programmed.
```

hathach, 2018-07-30:
Yeah, right. I changed it from dfu-boot to dfu-bootloader to be consistent with the "bootloader" only target that need jlink. But forgot to update the document. After the next usb PR, I will focus on bootloader, making it submodule + various enhancement, then I will update bootloader document for both nrf52832 and nrf52840 (e.g how to enable and use OTA etc ...)

hathach, 2018-07-30:
PS: and also migrating the bootloader on nrf52832 to S132 v6 as well, having the same API with S140 v6 will make maintaining repo easier.

dhalbert, 2018-07-30:
@hathach You beat me to it! https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/12

hathach, 2018-07-30:
Just 30 seconds before your comment 😉😉

hathach, 2018-08-15:
having trouble with S140 v6.1.0, couldnt start advertising, same code work with S140 v6.0.0. Nordic release note say the 6.1.0 is binary compatible with v6.0.0, seems there is some hidden issue. Posted in the devzone for support already.
https://devzone.nordicsemi.com/f/nordic-q-a/37473/advertising-does-not-start-with-s140-6-1-0-same-hex-run-without-issue-with-s140-6-0-0

hathach, 2018-08-23:
I think it is ok to close this now

ladyada, 2018-08-29:
ok i can also update/add nrf52832 details to this guide https://learn.adafruit.com/circuitpython-on-the-nrf52 - @dhalbert are we good to close?

dhalbert, 2018-08-29:
Sure, closing.

closed 2018-08-29 by dhalbert