adafruit/circuitpython #1070: nrf52 sys module does not have  stdin,stdout or stderr

jerryneedell, 2018-07-29:
This is related to the AMPY issue
[https://github.com/adafruit/ampy/issues/47](url)

the atmel samd and esp8266 builds have sys.stdout implmented, but the nrf52 build does not and tehampy "get" function fails since it relies on sys.stdout.

jerryneedell, 2018-07-29:
I noticed that some key items are not enable in the nrf mpconfigport.h

nrf mpconfigport.h
```
#define MICROPY_PY_SYS_STDFILES                  (0)
#define MICROPY_PY_SYS_STDIO_BUFFER              (0)

#define MICROPY_PY_IO                            (0)
#define MICROPY_PY_IO_FILEIO                     (0)
```

esp8266 mpconfigport.h

```
#define MICROPY_PY_IO               (1)
#define MICROPY_PY_IO_FILEIO        (1)

#define MICROPY_PY_SYS_STDFILES     (1)
#define MICROPY_PY_SYS_STDIO_BUFFER (1)
```

I tried enabling these, but the build failed:
```
jerryneedell@Ubuntu-Macmini:~/circuitpython_master/ports/nrf$ make BOARD=feather52832  SD=s132  SOFTDEV_VERSION=5.0.0 SERIAL=/dev/ttyUSB0 dfu-gen dfu-flash
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Create build-feather52832-s132/genhdr/pins.h
GEN build-feather52832-s132/genhdr/mpversion.h
GEN build-feather52832-s132/genhdr/qstr.i.last
QSTR updated
GEN build-feather52832-s132/genhdr/qstrdefs.generated.h
FREEZE freeze
LINK build-feather52832-s132/firmware.elf
build-feather52832-s132/py/modbuiltins.o:(.rodata.allowed_args.11668+0x14): undefined reference to `mp_sys_stdout_obj'
build-feather52832-s132/py/modsys.o: In function `mp_sys_print_exception':
modsys.c:(.text.mp_sys_print_exception+0x34): undefined reference to `mp_sys_stdout_obj'
build-feather52832-s132/py/modsys.o:(.rodata.mp_module_sys_globals_table+0x54): undefined reference to `mp_sys_stdin_obj'
build-feather52832-s132/py/modsys.o:(.rodata.mp_module_sys_globals_table+0x5c): undefined reference to `mp_sys_stdout_obj'
build-feather52832-s132/py/modsys.o:(.rodata.mp_module_sys_globals_table+0x64): undefined reference to `mp_sys_stderr_obj'
build-feather52832-s132/py/modsys.o:(.rodata.mp_sys_stdout_print+0x0): undefined reference to `mp_sys_stdout_obj'
build-feather52832-s132/py/modio.o:(.rodata.mp_module_io_globals_table+0x14): undefined reference to `mp_type_fileio'
collect2: error: ld returned 1 exit status
Makefile:340: recipe for target 'build-feather52832-s132/firmware.elf' failed
make: *** [build-feather52832-s132/firmware.elf] Error 1
```

perhaps I missed some  or there is some code missing from the nrf port. - I have not looked deeper yet.

tannewt, 2018-10-08:
@jerryneedell Is this done?