adafruit/circuitpython #1072: Using PulseIn to detect pin interrupts

notro, 2018-07-30:
I'm trying to emulate an [MCP23008](http://ww1.microchip.com/downloads/en/DeviceDoc/21919e.pdf) (I2C I/O Expander) and for that I need pin interrupts.

[pulseio.PulseIn](https://circuitpython.readthedocs.io/en/3.x/shared-bindings/pulseio/PulseIn.html) almost gives me what I need. 
All I need to know is:
- has an interrupt happened since last time I checked
- what's the current pin value.

These are the problems I face with PulseIn:
- I don't get access to the very first edge, so I loose the first interrupt.
- If the pin is active on init I loose another edge due to idle_state
- In case of jitter, the pulse buffer is easily overrun. I will know that there was an interrupt, but I don't know the current pin value so I have to check with DigitalInOut. This means that I loose out on the next edge.

Adding an edge counter to PulseIn would solve these problems (counter will be cleared on read).
It would also be nice to have a ```PulseIn.pinvalue``` or something.

Or would it be better to just implement proper pin interrupts?

dhalbert, 2018-07-30:
@tannewt and I discussed this. He suggested that we add a `.value` getter property on PulseIn, so you could just read the current value. We might read the pin directly, keep track of the value in the interrupt handler, or calculate the value based on the count. Does this sound like it would solve your problem?

We would like to implement interrupts, but doing interrupt handlers is a big topic (storage management, etc.). There's the MicroPython model but we'd like something more beginner-friendly.

notro, 2018-07-30:
Yeah I think a value property could do it. If I haven't seen any pulses since init, I can just check the value to catch that first edge.

I forgot about pull up, can we add a pull property as well or a kw arg to the contructor?

tannewt, 2018-07-30:
I don't think the samd can pull up when EXTI is active because the pinmux is enabled.

notro, 2018-08-06:
It looks like @BitKnitting makes it work:
https://github.com/BitKnitting/wakey_circuitpython/blob/master/src/lowpower.c#L52-L55

tannewt, 2018-10-08:
@notro I'm skeptical the pull up will actually work. Feel free to try it though.

gpshead, 2020-10-19:
Wouldn't a `.value` property on `PulseIn` still leave you with a race condition?

dhalbert, 2021-12-24:
`countio` seems like the better option for this now. Some `async/await` support would be good too. See #5606 for countio suggestions.

closed 2021-12-24 by dhalbert