adafruit/circuitpython #1083: Neopixel data timing too slow on Samd51 M4 boards

Thomasinator, 2018-08-03:
I discovered a problem driving a neopixel strip from adafruit. Whatever color, even all off, i send to the strip. The leds appear in a flickering full white. Driving a neopixel ring with the same script works fine.

I have an ItsyBitsy M4 with CircuitPython 3.0.0 installed and the Library bundle 3.x 20180802 on it.
I used the example from [here](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/CircuitPython_Essentials/CircuitPython_NeoPixel.py)

At work i could measure the output on D5 with an oszilloskop and the output is at a frequency at around 280 kHz. Normally it should be at 400 or 800 kHz.
Driving it with an Arduino UNO and the C-Library for NeoPixel from Adafruit produces an Output of 800 kHz and the Strip is working fine.

In the descriptions of the strip and the ring i have found only one difference, the ring could be with WS2812 or SK6812, the strip is only made from SK6812.
So it could be i have a ring with WS2812 and they work with the 280 kHz signal fine, and my strip is from SK6812 and won't work with the 280 kHz signal

So i think something is wrong with timing the output, or processor is running at lower frequency than it should.

Or am i make something wrong?

Adafruit NeoPixel Digital RGBW LED Strip - Black PCB 60 LED/m
PRODUCT ID: 2837 
NeoPixel Ring - 12 x 5050 RGBW LEDs w/ Integrated Drivers - Natural White - ~4500K
PRODUCT ID: 2852

Thomasinator, 2018-08-03:
Thanks,
i have considered this an initialized it with bpp=4 and pixel_order=(1,0,2,3).
But that does not explain why the frequency of the data is around 280 kHz

Thomasinator, 2018-08-03:
i have also tested this example form here
http://circuitpython.readthedocs.io/en/latest/shared-bindings/neopixel_write/__init__.html#module-neopixel_write

And there is also an output freqency of 280 kHz, seems to me something is wrong with the basic timing generating the output for the neopixel

jerryneedell, 2018-08-03:
Hmmm. I’ll try to reproduce this later today.

jerryneedell, 2018-08-03:
Did you try pixel_off=(0,0,0,0)

Thomasinator, 2018-08-03:
Yes,
i have changed the example to toggle between green (0,255,0,0) and off (0,0,0,0) every second.
But the strip keeps beeing something full white flickering

jerryneedell, 2018-08-03:
FYI -- I connected an 8 LED strip https://www.adafruit.com/product/2869  to my itsybtsy m4 express and I do see the 280kHz data rate, but thes trip works fine on bit my itsybitsy m4 and feather m4 -- note that this test 
```
import board
import neopixel_write
import digitalio
import time

pin = digitalio.DigitalInOut(board.D13)
pin.direction=digitalio.Direction.OUTPUT
pixel_off=bytearray([0,0,0,0])
pixel_on=bytearray([0,255,0,0])
while True:
    neopixel_write.neopixel_write(pin,pixel_off)
    time.sleep(1)
    neopixel_write.neopixel_write(pin,pixel_on)
    time.sleep(1)
```
toggles the first pixel on the strip RED (not green) but the doc indicates that ignores the pixel ordering, so I think this is expected.

jerryneedell, 2018-08-03:
also - FYI - I am powering the strip from the Itsybitsy 3V output and have it connected to D13.

jerryneedell, 2018-08-03:
ah -- interesting -- I can reproduce your issues if I power the strip from VUSB and use D5 for the data signal!!

dhalbert, 2018-08-03:
Thanks for detecting this problem. We'll look at the timing code. There are some loops that were calibrated to give the correct timing, but it seems like something is wrong with the measurements. Also, there is a instruction/data cache on the M4, and there may be an issue with its state when the `neopixel_write` code runs.

jerryneedell, 2018-08-03:
sorry - hit wrong button -- did not mean to close.  
@dhalbert   Any idea why it would depend on the voltage level used.

dhalbert, 2018-08-03:
might be waveform shape, rise/fall time on the level converter vs not

jerryneedell, 2018-08-03:
OK -- there are some big negative spikes when using 5V taht are not there at 3V.   It's not clear to ne that the data rate is the issue. Its something to look at , but may not be related to the performance.  My quick glance at the neopixel data sheet referred to maximum rates, but was less clear about minimum rates.

tannewt, 2018-08-08:
@Thomasinator could you test a new build from here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/itsybitsy_m4_express/ Its commit 2e80.

Thomasinator, 2018-08-08:
I'll try to test it the next few days

Thomasinator, 2018-08-09:
I did some measurements, the frequency is now 632 kHz for sending an OFF (0,0,0,0) to a pixel and 700 kHz for sending full ON (255,255,255,255). The example with green blinking is working for the ring but not for the strip. To supress the positiv/negativ spikes i use a 150ohm series resistor, the signal is clear as of a 3,3V digital output pin but with the 5v level.

Here the measured timing and the deviations to the datasheet timings
```
             WS2812  SK2812
T0H   480ns  + 80ns  +180ns
T1H   780ns  - 20ns  +180ns
T0L  1100ns  +250ns  +200ns
T1L   620ns  +170ns  + 20ns
```

dhalbert, 2018-08-09:
Yes, we recommend that the data signal voltage match the supply voltage, and use a level converter if necessary. I don't think what you're seeing is a timing issue now - it's related to trying to use the devices at the spec margins. That's one reason we added a 5V logic output pin to the Itsy Bitsy boards.

tannewt, 2018-08-10:
Its weird that its still not working for you. I reproduced the problem and it went away when I tightened up the timings. I don't expect it to be 800khz though since its being bitbanged.

Would you mind experimenting with the numbers yourself like I did in https://github.com/adafruit/circuitpython/commit/b0dd645e27219f8a593f7c5af1339cebffeebfe8 ?

Thanks!

Thomasinator, 2018-08-16:
Sorry, I don't know how to do these changes and get an .uf2 file to update my ItsyBitsyM4.

jerryneedell, 2018-08-21:
PR #1132 worked on my ITSY BITSY M4 express -- but I did find that I had to improve my wire connections to get it to work reliably.

PR #1096 (2018-08-08): Retune neopixel timings on SAMD51. They were too slow.
Test the timings a reference PWM output because CPU clock isn't perfect.

```python
import neopixel_write
import digitalio
import time
import board
import pulseio

pwm = pulseio.PWMOut(board.D10)
pwm.duty_cycle = 2 ** 15

pin = digitalio.DigitalInOut(board.D12)
pin.direction=digitalio.Direction.OUTPUT
pixel_off=bytearray([0,0,0,0])
pixel_on=bytearray([0,255,0,0])
while True:
    neopixel_write.neopixel_write(pin,pixel_off)
    time.sleep(1)
    neopixel_write.neopixel_write(pin,pixel_on)
    time.sleep(1)
```

Fixes #1083
commit b0dd645e (2018-08-08): Retune neopixel timings on SAMD51. They were too slow.

Fixes #1083
PR #1132 (2018-08-21): Speed up zero neopixel pulses.
SK6812 on 5v is pickier than WS2812 on 5v.

Hopefully fixes #1083.
commit 44de3d49 (2018-08-21): Speed up zero neopixel pulses.

SK6812 on 5v is pickier than WS2812 on 5v.

Hopefully fixes #1083.