adafruit/circuitpython #1084: Change main filename to run next

tannewt, 2018-08-03:
Add the ability to set the next main filename before triggering a reload. This can be used to switch between games on a system or between memory heavy animation scripts. https://forums.adafruit.com/viewtopic.php?f=60&t=139041

kevinjwalters, 2019-04-05:
My similar discussion: https://github.com/adafruit/circuitpython/issues/1084

robertgallup, 2020-08-27:
Mentioned this awhile ago on Discord, but thought I'd chime in here.

I'm working on a tutorial and have a bunch of example scripts all stored on the CIRCUITPY drive. Switching between them requires manually renaming or copying files to code.py/main.py. This gets tedious so I wrote a menu script that lists the available other scripts on the CIRCUITPY drive. Selecting one runs it using Import. This works, in general. The problem is that larger scripts can overflow the heap when imported even though they work fine on their own.

There might be a more general solution, but for me, it would be great to have something like sys.exit(_'next.py'_) that would restart the vm with the named script. No need for this to disrupt any existing behavior. e.g. ctrl-D from the REPL and sys.exit() could still load code.py/main.py by default.

The way this would work for my case is that the menu would be code.py/main.py and use sys.exit('next.py') to run the chosen script. All the example scripts would act normally, you'd use ctrl-C,ctrl-D to restart the menu, no special hooks to get back to the menu. Of course, if you wanted to chain several scripts, each one could use sys.exit(_'next.py'_).

deshipu, 2020-08-27:
I have exactly the same use case for my game consoles.

dhalbert, 2020-08-27:
We would not enhance `sys.exit()` for that purpose, because we aim to be a subset or the same as the CPython version. But we could certainly add it to `supervisor`, such as adding an arg to [`supervisor.reload()`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/supervisor/index.html?highlight=supervisor#supervisor.reload).

robertgallup, 2020-08-27:
Adding it to supervisor.reload() would be perfect.

tannewt, 2020-08-28:
Note that the hardest part of this is storing the filename because the heap gets reinitialized before running the file.

robertgallup, 2020-08-28:
Good point. Is this same issue handled in `supervisor.set_next_stack_limit()`? Or, is it a different problem?

tannewt, 2020-08-28:
Ya, it is similar but stack size is a fixed size so it's easy to just statically allocate it. We could simply have a char* array with a limit to start. If folks hit the limit often we could either increase it or make the allocation dynamic.

robertgallup, 2020-08-28:
Ah ha. I see the issue. Picking a good limit is key (though, no matter what you pick, someone will have a use case for more :) ).

dhalbert, 2020-08-30:
If the heap memory is not touched when the VM shuts down, then I think the following would work. It relies on being to able to use the stale data in the old heap.

1. Copy the location of the string object that is the filename into a pointer variable at a fixed location
2. Reload exception throws us back to somewhere in main.
3. VM gets shut down. Loop back to restart VM
4. Before heap is re-initialized (is it zeroed?), examine the string object pointed to. Allocate an array on the stack large enough to hold the filename, and copy the filename to there.
5. Initialize the heap.
6. Pass the filename on the stack to `run_code_py().

deshipu, 2020-08-30:
I just realized there is one more piece of the puzzle missing — after the code finishes running, it would be best if the device restarted again, with the default `main.py` this time, instead of dropping into the REPL. I wonder if this is too much to ask for an obscure use case, though.

dhalbert, 2020-08-30:
@deshipu If each of the alternate programs did a `supervisor.reload()` with no args when it's done, that would just happen. Are you thinking that is an extra piece of code that the alternative programs wouldn't have?

deshipu, 2020-08-30:
Yes, ideally it would be the same kind of programs that you normally run as `code.py` — whatever the users have written or copied from examples. Requiring a reset at the end is a possible solution, but I'm sure it will often be missed and users will wonder why the device is not displaying the menu again after running their code.

dhalbert, 2020-08-30:
[started to reply and then saw your edited reply]

It would be possible to set a flag that says to reload on exit, either via `supervisor.reload()` or some new function. But as I was going to mention, and you already did, most example programs already are infinite loops and have no exit mechanism, so they'd have to be modified anyway.

deshipu, 2020-08-30:
Also, you probably don't want to restart if there was an error, only if it was a normal end of the program. I might be overthinking this.

tannewt, 2020-08-31:
> Also, you probably don't want to restart if there was an error, only if it was a normal end of the program. I might be overthinking this.

I don't think you are overcomplicating it. I think it is complicated.

Another aspect is what file should be run on reload. We don't know what file has been changed when the FS is written. My guess is that we want the currently running file reloaded (and therefore we need to preserve the filename past start up into the new heap.)

A related use to this would be doing OTA file updates over wifi. We'll want a known good code.py to allow for writing new files even when the loaded file crashes.

robertgallup, 2020-08-31:
So, I might be misunderstanding, but are you suggesting that `supervisor.reload('file_name')` would permanently override code.py/main.py as the file for reload? Or, are you talking about behavior on the next reload after an error? Or, neither of those...

tannewt, 2020-09-01:
I think we'll want a separate function rather than putting it in `reload`. That way the code can set if for all reload causes, not just manual reload. Something closer to `set_next_stack_limit`: https://circuitpython.readthedocs.io/en/5.3.x/shared-bindings/supervisor/__init__.html#supervisor.set_next_stack_limit

Then we can have multiple kwargs for different reload reasons or separate functions. I imagine you want: reload, exception, safe mode and completion as reasons.

robertgallup, 2020-09-02:
This sounds like a good general approach. As I envision my current use case, I just need to reload with a specific file without affecting any of the defaults. Though, with more options, my use case might evolve.

deshipu, 2020-09-02:
After some thought, I think that just setting the file name for the next run would solve 90% of m case as well — the users always can reset the device to get back to the menu, and an "exit the game" option can be provided by the library, which would then do the correct thing.

siddacious, 2020-09-08:
My loader for the OSHWA 2020 badge would appreciate this functionality. I assume that since the proposed mechanism is simply designating which file to run, there would be no need to make sure pins are released/de-initialized?

As you suggested previously @tannewt , this will mean we (I ) will need to handle the "don't reinit already initialized stuff" case for libraries. Since the libraries will need to know the 'commanded reload' vs 'default start' state, it would/could be exposed by`supervisor` or similar?

tannewt, 2020-09-08:
@siddacious I was thinking it would be a full reload that would reset everything.

What is an example of things you don't want to reset?

robertgallup, 2020-09-09:
For my use case, I think a full reload/reset would be optimal.

siddacious, 2020-09-11:
@tannewt I was referring to our previous convo where you suggested thinking about how sensors libs should handle a restart, though in that context I think you were talking about waking up from a sleep that was entered for power saving purposes so I may have crossed the streams.

It sounds like this is a related but different issue as there are certainly use cases where you _would_ want a complete reset, sensors included ie: running an app that wants different configurations for a sensor.

robertgallup, 2020-09-16:
One thing that just occurred to me is that, if possible, it might be useful to support `.mpy` files in this function. Currently, `.mpy ` isn't supported during a board restart.

dhalbert, 2020-09-16:
@robertgallup I am not sure what you mean. Do you mean you can't have code.mpy?

If something.py and something.mpy both exist, then something.py will take precedence.

deshipu, 2020-09-16:
I think I tested it a while ago, and indeed `code.mpy` or `main.mpy` weren't picked up.

dhalbert, 2020-09-16:
We should fix that, but it's easy to have a code.py that imports the actual program.

robertgallup, 2020-09-16:
@dhalbert @deshipu Yes. I haven't tested `code.mpy`, but `main.mpy` is not recognized. There was a discussion about this on Discord some time back.

deshipu, 2020-09-16:
Sure, but it's always some extra memory, and the reason for using mpy is to save memory in the first place.

robertgallup, 2020-09-16:
And the reason for this restart() function is to avoid using import to begin with.

dhalbert, 2020-09-16:
I think it's just a matter of adding code.mpy and main.mpy to:

robertgallup, 2020-09-16:
*That* would be a simple fix for the standard restart. :-) With this new function, you'd either have to accept the full file name, e.g. `cool_program.mpy' or make sure to check for the `.mpy` version (I guess the same way that `import` currently works.)

tannewt, 2020-09-16:
I'd rather not support mpy at the top level because it obfuscates the running code and is impossible to edit.

robertgallup, 2020-09-16:
A fair point. But, I'm curious, why is non-editable, obfuscated code undesirable? Are you worried about nefarious applications? Or, is it against the hack/education ethic of CircuitPython?

tannewt, 2020-09-16:
It's against the open, accessible philosophy of CircuitPython.

The only advantage of it is reducing memory overhead during load. Otherwise, plain .py should be used.

Some folks think the obfuscation provides a level of secrecy of their code but that's more of an illusion than reality. All variable names are preserved and no optimization is done on the byte codes that I know of. So, it'd be straightforward getting something like the original code back. Only comments would be missing.

robertgallup, 2020-09-16:
Agree. Having all code as `.py` would be optimal. But, since microcontrollers are resource constrained there are reasons for using `.mpy` files (e.g. the library bundle). And, once CircuitPython allows `.mpy` for modules, you can have a top level `main.py` that just imports a `.mpy` and it seems like you're back to code that's not open and accessible. So, not allowing `.mpy` at the top level doesn't seem to gain much and creates an artificial inconsistency. FWIW

tannewt, 2020-09-17:
I'd like to move away from mpy in the long term. Allowing code.mpy will make that harder.

robertgallup, 2020-09-17:
Makes sense. And, as you point out, there is very little real advantage for `.mpy` at the top level. And, it's not required for my use case.  :-)

cwalther, 2020-09-20:
I have written a prototype implementation of what is being discussed here at [cwalther:setnextcode](https://github.com/cwalther/circuitpython/tree/setnextcode) (cwalther@f222a3f735f12f8fb630d700e408189cffc0669e). Not necessarily meant to be taken as-is (hence not a pull request yet), but just to have a basis for discussion. Maybe someone else is already working on it as well and we can combine our ideas. It covers the cases we need for pewpew-game/game-m4-menu#1, but maybe others have additional requirements. Reviews also appreciated because I am very unfamiliar with CircuitPython/MicroPython internals, so there may be better ways to do things.

tannewt, 2020-09-21:
@cwalther Would you mind doing a draft PR? PRs are much easier to comment on.

cwalther, 2020-09-22:
Done!

I also extened it to accept `None` or an empty string to reset to the default search order, which may be useful when a program that was itself started as an alternative next code wants to reboot into the standard code.py/main.py rather than itself using `supervisor.reload()` (such as https://github.com/python-ugame/circuitpython-stage/blob/9596a5904ed757e6fbffcf03e7aa77ae9ecf5223/pewpew_m4/ugame.py#L39), and may make it unnecessary to add a distinction between a reload from `supervisor.reload()` and one from USB filesystem writes.

cwalther, 2020-09-25:
Based on @tannewt’s first comments on #3454, let’s collect some more detailed requirements.

My work so far has concentrated on the **game console** use case, which I envision as follows (@deshipu or @robertgallup, correct me if you have any additions):

* There is a menu, main.py (written by us experts), and a number of games: game.py, game2.py, … (written potentially by beginners).
* Turning on the device boots into the menu, which lets the user choose one of the games.
* Choosing “game.py” reboots into game.py.
* When game.py
  * is interrupted by a USB file write: reboot into game.py.
  * exits
    * successfully: immediately reboot into main.py.
    * with an exception: wait in the “Press any key to enter the REPL. Use CTRL-D to reload.” state so that the user can see the traceback.
      * When reloaded by USB file write, reboot into game.py. (This is not correctly handled by my current prototype, it reboots into main.py.)
      * When reloaded by CTRL-D immediately or CTRL-D in the REPL, reboot into either main.py or game.py, probably doesn’t matter which.
* Games should not need to do anything with `supervisor` or any other special action to exit “properly” and support the above workflow, they should just be able to run off the end or call `sys.exit()`.
* While developing a game, you can name it “code.py” to bypass the menu, afterwards you rename it to “game.py”.
* The same game.py should also work on a MicroPython device that does not have `supervisor` (with a different main.py and different implementations of the libraries it uses) in a more traditional run-by-import workflow.

What would the **OTA update** use case look like? Something like this?

* code.py receives an updated version of itself and saves it as new_code.py.
* When it determines that the download has been successful (perhaps with a CRC check), it reboots into new_code.py.
* When new_code.py
  * exits with an exception: reboot (immediately or with waiting?) into code.py. code.py should be able to retrieve what the exception was and tell the user that the update failed.
  * gets this far without an exception: renames itself to code.py and reboots into code.py.
* What if more modules than just code.py need to be updated?

Are there any other use cases that are fundamentally different from the above?

Apart from retrieving an exception from the previous run, which seems like a somewhat separate feature, I see two areas where the requirements of the different use cases may differ:

**A.** After next code exits without having called `supervisor.set_next_code_file()`, do we preserve the next code info (next reload goes into the same file again) or clear it (next reload goes into code.py/main.py)?

.  | Game Console | OTA
-- | ------------ | ---
when calling `supervisor.reload()` | either | either
when interrupted by a USB file write | preserve | preserve
when completed with an exception | either | clear
when completed successfully | clear | either (never happens in the above workflow)

**B.** After next code exits, do we wait in the “Press any key to enter the REPL. Use CTRL-D to reload.” state or reload immediately?

.  | Game Console | OTA
-- | ------------ | ---
when calling `supervisor.reload()` | reload | reload
when interrupted by a USB file write | reload | reload
when completed with an exception | wait | ?
when completed successfully | reload | either (never happens in the above workflow)

My current prototype has B partially configurable using the `reload_on_success` keyword argument, but not A. Can we fill in the question marks or fix mistakes in the tables above to figure out if and how much either of them needs to be configurable? Or do we just decide to make them maximally configurable anyway? So far it seems like we never need to distinguish between the first two rows and never need them configurable. The last two rows are less clear. (I would like to have B configurable in the “successfully” row even if the table indicates that it wouldn’t need to be, because the behavior required by the “game console” use case there is different from the behavior after code.py/main.py after a hard reboot, and some use case we haven’t envisioned yet might want it the same.)

robertgallup, 2020-09-25:
Great requirements spec! A couple of things caught my attention.

1) What's the use case for the requirement to reboot into game.py on a USB file write (in the case A table, above) ? (I'm not sure this is necessary/desirable for my use case -- but, possibly because I'm overlooking something.)

2) Exception handling. For my case, if "game.py" is running and I want to stop it and return to the main menu, I would expect to use CTRL+C, CTRL+D. From my read, it sounds like you're proposing this could return to either game.py or the main menu? I think my use case would require returning to the main menu.

deshipu, 2020-09-25:
> What's the use case for the requirement to reboot into game.py on a USB file write (in the case A table, above)? (I'm not sure this is necessary/desirable for my use case -- but, possibly because I'm overlooking something.)

If you are working on your game, and modifying the file, you want to see your changes taking effect immediately without having to select it from the menu each time.

cwalther, 2020-09-25:
> For my case, if "game.py" is running and I want to stop it and return to the main menu, I would expect to use CTRL+C, CTRL+D. From my read, it sounds like you're proposing this could return to either game.py or the main menu? I think my use case would require returning to the main menu.

Good point, I’ve changed that.

robertgallup, 2020-09-25:
> If you are working on your game, and modifying the file, you want to see your changes taking effect immediately without having to select it from the menu each time.

OK. I see. To me, this seems like a special case that might lead to unintended consequences/confusion in other applications and might be better handled by a developer disabling or bypassing the menu during development (e.g. renaming "game.py" to "code.py"). But, that's from my particular perspective.

cwalther, 2020-09-25:
Can you think of a concrete example of a situation where it would be unintended or confusing?

I don’t have a strong opinion either way, I guess I tend toward @deshipu’s view because I’m working on his devices, and also @tannewt has said above
> My guess is that we want the currently running file reloaded

robertgallup, 2020-09-25:
> Can you think of a concrete example of a situation where it would be unintended or confusing?

I might be misconstruing this, but let's say that I'm working on a suite of programs. I might be making a change to a file other than the one that's currently running. In this case, I'd rather re-run from the top. Or, I might be adding a new game and want to see if the menu picks it up properly. Again, I'd want to rerun the menu. Trying to infer developer intent from a USB write can be difficult.

If the case were changed to "if game.py is running and game.py is updated, then rerun game.py" it might have less potential for interference.

In any case, I don't think this breaks my use case, it's just an observation that this is an exception to the general rule that currently controls autorun ... and, exceptions like this make me nervous.  :-)

cwalther, 2020-09-25:
Thanks, that helped me understand why you see it as an “exception” or “special case”. I was operating under the assumption that “the general rule that currently controls autorun” was equivalent to “always rerun the file that’s currently running”, because there was only one file that that could be, so it wouldn’t be an exception. But now I see that that’s not true: If you only have main.py and main.py is running and then you save a new code.py that didn’t previously exist, it will run code.py.

Food for thought. Any other opinions?

> If the case were changed to "if game.py is running and game.py is updated, then rerun game.py" it might have less potential for interference.

Agreed, but as far as I understand, that’s not possible, or at least not without a lot more code.

deshipu, 2020-09-25:
You also would get surprising behavior when you update a file that is imported in your `game.py` but not run directly.

deshipu, 2020-09-25:
The way I see it, the easiest to understand and explain behavior (apart from the current "always run `main.py` or `code.py`") would be:
- if a program finished running, or there was an error (including `KeyboardInterrupt`), go back to the default behavior of running the `main.py` or `code.py`, and, depending on the options, show REPL, exception message with a pause, or reset immediately.
- if there was a filesystem change, reload the currently running program.

tannewt, 2020-09-25:
>     * with an exception: wait in the “Press any key to enter the REPL. Use CTRL-D to reload.” state so that the user can see the traceback.

I'm not sure you actually want to ever enter this state because in the game console case you may not be connected to a host computer. I suppose you could reset the device to get out of it though. An alternative would be to have another `.py` or `main.py` to handle exceptions, kind of like a 404 page. That way it can be branded and also read inputs like buttons to go back to the menu. (CircuitPython doesn't know about these.)

Let's not worry too much about the OTA case at this point. It has separate issues because we may want to assume not having access to the device. I think the game console model will be enough to experiment with. (main.py is the bootloader and is never replaced. the user code can never finish if it wants to update itself another way.)

cwalther, 2020-09-25:
> I'm not sure you actually want to ever enter this state because in the game console case you may not be connected to a host computer.

But then, if you don’t happen to be connected to a host computer, you can’t see the traceback (on the display of the device) and don’t know what the bug in your game was. (The game console use case I have in mind is more aimed at (beginner) game developers than game players. They may be “playing”, not “developing” the game at this moment, but still be interested in identifying bugs when they happen.)

> An alternative would be to have another `.py` or `main.py` to handle exceptions, kind of like a 404 page. That way it can be branded and also read inputs like buttons to go back to the menu. (CircuitPython doesn't know about these.)

That’s an interesting idea. I’ll experiment with that.

---

In the meantime I have gone ahead and made it “maximally configurable” for prototype testing: added keyword arguments for the rows (rows 1 and 2 combined = “on reload”, row 3 = “on error”, row 4 = “on success”) of both tables (A = “sticky”, B = “reload”). [Documentation here.](https://github.com/cwalther/circuitpython/blob/c97078dd841798c3c1c8b45370cabb4656c9f8da/shared-bindings/supervisor/__init__.c#L113)

When it turns out that some of these options are never used, always used with the same value, or always used together, we can then remove or combine them. My current expectation is that `reload_on_success` will stay, `sticky_on_reload` (the controversial one from the last few posts) will stay if we can’t agree on a default value for it, and the rest can go.

So my question for interested parties is: Using these options, can you make it work the way you need for your use cases? And if yes, with what values?

My answer for the PewPew game console case is “yes, reload_on_success=True, sticky_on_reload=True (preferred, not absolutely requried)”.

---

I notice we are at 6.0.0-beta.0 now. What are the chances of getting this into 6.0? I have no particular preference, just to set expectations. There is one commit waiting at https://github.com/python-ugame/circuitpython-stage already (submodule `frozen/circuitpython-stage`) that should go into 6.0 in my opinion. Once this lands, there will be more.

deshipu, 2020-09-25:
Another option would be for the menu program to check for exception and display it (or otherwise handle) in a user-friendly manner. This way we can use the console's button to scroll and dismiss the message, for example.

cwalther, 2020-09-26:
Right, I also came to that conclusion when I was mulling it over under the shower. That actually seems the more elegant solution to me. The ability to retrieve the last exception is required either way.

So until someone convinces me otherwise, I think there is no need to have different next file names for different outcomes. The capabilities of the current prototype plus retrieving the last exception are sufficient.

deshipu, 2020-09-26:
So all we need is an option that says what happens when the program finishes — either show the exception and drop to REPL (the default), or get back to the main program, possibly with a stored exception, and let that program decide what to do next.

cwalther, 2020-09-26:
Translated into prototype-speak, that option would be a combined `reload_on_error = reload_on_success` with a default of `False`, while the rest would be fixed at `sticky_on_reload=True`, `sticky_on_error=False`, `sticky_on_success=False`, if I understand you correctly. Works for me.

However, for now, until we have established that the “retrieve last exception” feature is feasible and can land at the same time (which I intend to try, but no promise), I would prefer to keep `reload_on_error` and `reload_on_success` separate, because without that feature we need `reload_on_error=False` and `reload_on_success=True`.

robertgallup, 2020-09-26:
If I'm reading this correctly, I think for my case, all of the `sticky_on_...` flags would be `False` which looks like the default? So, in my main menu, to jump to a menu choice, I would use:

```
supervisor.set_next_code_file(filename='example.py')
supervisor.reload()
```

Is that correct?

And, if I press the reset button or disconnect/reconnect the device, everything returns to default, running `code.py` or `main.py`?

cwalther, 2020-09-26:
If you’re okay with staying in the “Code done running. Waiting for reload. / Press any key to enter the REPL. Use CTRL-D to reload.” wait state after example.py has finished, that is correct, except that the `filename` argument is positional-only (that is not deliberate, but I didn’t see any way of making it positional-or-keyword).

What I do is
```python
supervisor.set_next_code_file(selected + '.py', reload_on_success=True, sticky_on_reload=True)
supervisor.reload()
```

The `reload_on_success=True` makes it skip the wait state after finishing without error and go right back to the menu, and the `sticky_on_reload=True` is our difference of opinion on what a USB file write should do.

And yes, a hard reset returns everything to default.

[18 more replies]

closed 2021-06-25 by hierophect