adafruit/circuitpython #1090: Initial pop sound when playing samples via audioio.AudioOut

CedarGroveStudios, 2018-08-06:
When using ``audioio``, I noticed that the DAC output makes a loud pop noise when it starts playing a raw sample or wave file. I suspect (but haven't verified) that it happens because the initial DAC value before playback is zero whereas the sample's zero voltage value correlates to the mid-value of the DAC. Playing the sample/file causes a large output voltage jump that, if the output pin is decoupled through a series capacitor, creates a noticeably loud pop.
Since ``audioio`` is used for an AC audio stream with an zero baseline offset equal to 1/2 of the DAC's full scale, I'm wondering if it would be possible to slowly precondition the DAC output to the mid-point prior to playing a sample or wave file, then start playback with the next value in the stream that equals the mid-point value (within a reasonable threshold) -- in other words, manage the audio playback with reference to the sound's zero-crossing points.
This would primarily affect the start and stop of playback. Pause and resume should not be impacted since the DAC playback value should be frozen at a level close to the next value.

CedarGroveStudios, 2018-08-06:
This was experienced using the Essentials learning guide's Play A Tone example on a Feather M0 Express running CircuitPython 3.0.0.

ladyada, 2018-08-06:
@CedarGroveStudios yah...what happens if you try to 'diy' with an analog write before playing a file?

CedarGroveStudios, 2018-08-06:
Haven't tried that yet. Was trying to decode the DAC initialization routines in the library first to see if the DAC is always set to zero value before playback. Decoding libraries is not my strong suit so I decided to raise the issue for the experts. I'll give your suggestion a shot late this afternoon (Pacific). BTW, I also encountered a similar pop when using the I2S DAC breakout board.

ladyada, 2018-08-06:
yeah,  for i2s its actually a side effect of the i2s clock stopping.
*ideally* once the i2s object is instantiated, it will continuously generate '0' data to keep pops from occuring, i had this happen in raspi as well!

CedarGroveStudios, 2018-08-06:
For audioio would it be possible to start the I2S clock (or precondition the DAC) when the library is first instantiated rather than at the beginning of playback?

ladyada, 2018-08-06:
we probably should, @tannewt want an issue for ^?

CedarGroveStudios, 2018-08-06:
For I2S, is the data signed or unsigned? Is a value of zero the signal's mid-point?

ladyada, 2018-08-06:
i2s is signed, so zero would be midpoint

CedarGroveStudios, 2018-08-06:
Would it make sense from a consistency point-of-view to change auduoio to respond to signed values around zero for the DAC and PWM modes, too?

ladyada, 2018-08-06:
nope - id like to keep it as is, to 'mirror' how ADC works. DACs don't go negative, they start at 0 (MIN) and end up at 3.3V (MAX). also, dacs can be usd for non-audio things. 
i cant find the reference now about i2s being signed. either way, it would be set to midpoint.

CedarGroveStudios, 2018-08-06:
Understand. Will be easy for me to create a helper for my synth projects that adjusts and normalizes incoming and stored signed streams to unsigned DAC values for audioio.

dhalbert, 2018-08-06:
I was worried that the DAC(s) can be shared while in use, but for AudioOut it claims them for exclusive use. We could ramp up the output over a certain period of time.

Are there use cases where you don't want full-scale output, so the logical "zero" point is not 0.5*3.3V? E.g., you might want 0.0-1.0V output. Or is it always AC-coupled? And if that's specifiable, should the ramp-up time also be specifiable? What's a good value for that?

dhalbert, 2018-08-06:
I2S is signed, according to Wikipedia, which cites the original spec: https://en.wikipedia.org/wiki/I%C2%B2S

CedarGroveStudios, 2018-08-06:
For my applications, since audio out is AC coupled anyway, I would prefer to have the DAC's maximum resolution available for all streams. Mid point at 3.3 / 2 is perfect. Amplitude adjustment to 1v pk-to-pk would be done algorithmically to the data stream or stored sample prior to playback.

CedarGroveStudios, 2018-08-06:
As for the preconditioning delay, I'm guessing that keeping longer than 1/4 wavelength of 15Hz (> 16ms) should keep the ramp-up from passing through most decoupling capacitor arrangements.

CedarGroveStudios, 2018-08-06:
… although a volume setting attribute for AudioOut would be useful.

CedarGroveStudios, 2018-08-06:
@dhalbert  Since I2S is signed, that invalidates my previous test. I had assumed that AudioOut's data input range was consistent regardless of the output protocol (I2S was the same as DAC output). I'll rerun the I2S test with that in mind.

CedarGroveStudios, 2018-08-07:
**I2S Breakout Test:** Just completed the I2S test using the I2S Stereo Decoder - UDA1334A Breakout, CPy 3.0.0, and a Feather M0 Express using a modified version of the learning guide's example code. Confirmed that via ``AudioOut``, the input values are indeed signed with a min of -32767 and a max of +32767 around a baseline of 0. The data simply wraps if the value goes outside of that range rather than clipping (as it should). Playing all zeros (the baseline) through a ``RawSample`` array produced no audible clicks or pops. Also found that the breakout worked okay with a sample rate up to 44.1k samples/sec.

Playing and stopping seems to reset the DACs to the baseline value. Pause stops playback and holds the current DAC values. Resume picks up and plays the next value in the array. Pause and resume each generate a pop, that on the o'scope looked to me to be caused by the slowing decaying decoupling capacitor losing the DAC's held value. Scope pix are available if you need 'em.

In the process, I found a couple of errors in the CPy example for the breakout board. I'll work with Kattni to correct.

**Conclusion** is that the ``AudioOut`` library for I2S should be modified to clip at the minimum and maximum values rather than roll-over. Would likely benefit from DAC post- and pre-conditioning for short duty-cycle play/stop and pause/resume sequences. 

I'll test the native M0 DAC output next. After a dinner break.

CedarGroveStudios, 2018-08-07:
**M0 Express DAC Test:** Yes, ramping up to the midpoint level did help to quiet the initial pop sound when a `RawSample` array was played.

When `AudioOut` was first instantiated, it appeared that the DAC output remained at its previous value; 0 after a board reset or some other value based upon previous playback. If the DAC value was slowly increased to the mid-point value of 2^15 (over 30 ms) before `AudioOut` was instantiated, the initial pop sound was minimized when first executing `Play`. Subsequently, `Stop` froze the DAC at the current value which was then held until `Play` was executed again.

Since audio output will usually be capacitively-coupled, a slight popping sound may be heard when resuming playback after a `Pause`, due to capacitive discharge.

**Conclusion:** Incorporating a gradual ramp-up or ramp-down of the DAC to the midpoint value will help to reduce the popping noise during playback functions. The controlled ramp-up/down needs to happen when `AudioOut` is instantiated as well as prior to and after `Start`, `Stop`, and perhaps `Pause`, and `Resume` playback functions. It may also be useful to look at resetting DAC outputs to zero when `deinit` -ing a pin so that the DAC value prior to use is predictable.

tannewt, 2018-10-05:
I'm going to try and add this tomorrow. How many steps between zero and midpoint are needed during the 30ms ramp? I've gotta generate a sample for it under the hood. Should I ramp to the value at the start of the sample instead?

CedarGroveStudios, 2018-10-05:
Given the recommended playback of 22k samples/sec, that would be 660 sample steps minimum. That's ~2.5mV per step to reach a 1.65V midpoint.

A slow ramp from the current DAC output to the value at the beginning of the sample to be played is an excellent approach and anticipates poorly edited samples. I recommend keeping the ramp steps at no greater than ~2.5mV even though it could take up to 60ms in the extreme case.

Stop, Pause, and Resume would also benefit from the ramp approach. Resume is the more critical case since the typical decoupling capacitor will discharge after a Pause.

dhalbert, 2018-10-17:
I have a basic question after reviewing the PR, #1280. When I heard about doing this I thought you were going to doing the ramping based on the first and last values in a sample when you started and stopped playing that sample, so that no matter what was at the beginning or end of the sample, the output would smoothly ramp to match that. I read the issue in more detail and I realize it was not spec'd that way.  @CedarGroveStudios, is that impractical, or just not how it is done?

CedarGroveStudios, 2018-10-17:
Sorry that I didn't make that clearer in my description.
It's standard practice for studios to record samples with a zero value at the start and at the end so ramping up to mid-point should work for those. If we want to handle "non-standard" samples that may start or end at some value other than zero, then what you described would be the ideal approach. That method is particularly important when pausing and resuming mid-sample since the DAC's resting value would be pretty random.
-- There are lots of "non-standard" samples out in the wild. ;)   --
The other approach is to only start, stop, and resume playback when sample data is at zero or passing through zero (the zero-crossing approach). The drawback to the zero-crossing method is that it's very difficult to do with synchronized multiple tracks or stereo without creating phasing errors.

dhalbert, 2018-10-17:
Thanks, yes, I was thinking that most clips are probably not produced by someone familiar with the idea of inserting silence at the beginning and end.

closed 2018-10-17 by dhalbert

kevinjwalters, 2019-03-07:
What's the summary of the behaviour now for M0 and M4 boards?

If I do

> play whole sample 1
> play whole sample 2

and they both start/end with 32768 will it play them back to back without any additional delays? If sample 1 ends at 0 and sample 2 starts at 65535 how long will it take to ramp from one to the other and is this controllable from CircuitPython program?

For this scenario,

> play very short sample 1 looping
> stop sample 1
> play very short sample 2 looping

is there a way to control the bahaviour for **stop()** method on **audioio.AudioOut** to either immediately stop or stop at the end of the sample? The immediate approach leaves it in an unpredictable end state and for some applications it will be more desirable to complete the current sample, terminate the looping and then get to the next one.

tannewt, 2019-03-08:
The slew rate is not controllable from Python. It is [50 us per 49 DAC value](https://github.com/adafruit/circuitpython/pull/1280/files#diff-13dec12304b3d13dd106bfcf6c78160eR60).

I recommend samples start and end with 32768 for best results. The ramp will be zero length in that case assuming 32768 is also the quiescent value.

There isn't a way to stop after the end of the sample. That would be a great feature though so please file a separate issue.

kevinjwalters, 2019-03-08:
For future reference, the related forum discussion: https://forums.adafruit.com/viewtopic.php?f=60&t=148191

PR #1280 (2018-10-17): Ramp values to and from a default value while active.
This reduces the popping sound on initial playback of an audio
sample.

The M4 DAC has a pop on startup that cannot be prevented. It also
does not allow readback so current values of the DAC are ignored.

Fixes #1090
commit c209165d (2018-10-17): Ramp values to and from a default value while active.

This reduces the popping sound on initial playback of an audio
sample.

The M4 DAC has a pop on startup that cannot be prevented. It also
does not allow readback so current values of the DAC are ignored.

Fixes #1090