adafruit/circuitpython #1092: REPL: Ctrl-C doesn't raise KeyboardInterrupt

notro, 2018-08-06:
Pressing Ctrl-C when sleeping aborts the operation but doesn't raise an exception like in CPython.

CircuitPython:
```
Adafruit CircuitPython patchbase-3-gedb366e7f-dirty on 2018-08-06; Adafruit Feather M0 Express with samd21g18
>>> import time
>>> time.sleep(10)
>>>

```

CPython:
```
Python 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import time
>>> time.sleep(10)
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt
>>>

```

Another problem is that that the Ctrl-C event gets stuck so the next sleep is also aborted:
```
Adafruit CircuitPython patchbase-3-gedb366e7f-dirty on 2018-08-06; Adafruit Feather M0 Express with samd21g18
>>> import time
>>> t = time.monotonic(); time.sleep(1); print(time.monotonic() - t)
1.0
>>> t = time.monotonic(); time.sleep(1); print(time.monotonic() - t)
[[PRESS CTRL-C]]
0.333984
>>> t = time.monotonic(); time.sleep(1); print(time.monotonic() - t)
0.0
>>>
```

dhalbert, 2018-08-06:
Ctrl-C raises an interrupt in 2.x and 3.0.0, so this is something that has changed in master only recently.
```
Adafruit CircuitPython 2.3.1 on 2018-05-07; Adafruit Metro M0 Express with samd21g18
>>> import time
>>> time.sleep(10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
>>> 
```
```
Adafruit CircuitPython 3.0.0 on 2018-07-09; Adafruit Metro M0 Express with samd21g18
>>> import time
>>> time.sleep(10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
>>> 
```

notro, 2018-08-06:
Ok, I see that the pending exception isn't stuck either on a 3.x I have. The next ```time.sleep``` works just fine.

notro, 2018-08-17:
This differs between 3.x and 4.x:
```diff
diff --git a/py/vm.c b/py/vm.c
index 72d0d0d60..b5f53ee9a 100644
--- a/py/vm.c
+++ b/py/vm.c
@@ -136,7 +136,7 @@ mp_vm_return_kind_t mp_execute_bytecode(mp_code_state_t *code_state, volatile mp
     #define ENTRY(op) entry_##op
     #define ENTRY_DEFAULT entry_default
 #else
-    #define DISPATCH() goto dispatch_loop
+    #define DISPATCH() break
     #define DISPATCH_WITH_PEND_EXC_CHECK() goto pending_exception_check
     #define ENTRY(op) case op
     #define ENTRY_DEFAULT default
```

Making that change gives the expected exception:
```
Adafruit CircuitPython patchbase-dirty on 2018-08-17; Adafruit Feather M0 Express with samd21g18
>>> import time
>>> time.sleep(10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt:
>>>
```

tannewt, 2018-08-17:
@notro Any idea why? It looks like the change was intentional: https://github.com/micropython/micropython/commit/869024dd6e62905b7e1069b547856a769b3b24ba

notro, 2018-08-18:
Hmm, 2.5% perfomance improvement is nice.

Here are the various places where ```mp_pending_exception``` is checked:

[py/vm.c](http://elixir.tronnes.org/circuitpython/latest/source/py/vm.c):
```c
mp_vm_return_kind_t mp_execute_bytecode(mp_code_state_t *code_state, volatile mp_obj_t inject_exc) {
...
pending_exception_check:
                MICROPY_VM_HOOK_LOOP

                #if MICROPY_ENABLE_SCHEDULER
...
                #else
                // This is an inlined variant of mp_handle_pending
                if (MP_STATE_VM(mp_pending_exception) != MP_OBJ_NULL) {
                    MARK_EXC_IP_SELECTIVE();
                    mp_obj_t obj = MP_STATE_VM(mp_pending_exception);
                    MP_STATE_VM(mp_pending_exception) = MP_OBJ_NULL;
                    RAISE(obj);
                }
                #endif

```

[ports/unix/unix_mphal.c](http://elixir.tronnes.org/circuitpython/latest/source/ports/unix/unix_mphal.c):
```c
STATIC void sighandler(int signum) {
    if (signum == SIGINT) {
        #if MICROPY_ASYNC_KBD_INTR
        mp_obj_exception_clear_traceback(MP_OBJ_FROM_PTR(&MP_STATE_VM(mp_kbd_exception)));
        sigset_t mask;
        sigemptyset(&mask);
        // On entry to handler, its signal is blocked, and unblocked on
        // normal exit. As we instead perform longjmp, unblock it manually.
        sigprocmask(SIG_SETMASK, &mask, NULL);
        nlr_raise(MP_OBJ_FROM_PTR(&MP_STATE_VM(mp_kbd_exception)));
        #else
        if (MP_STATE_VM(mp_pending_exception) == MP_OBJ_FROM_PTR(&MP_STATE_VM(mp_kbd_exception))) {
            // this is the second time we are called, so die straight away
            exit(1);
        }
        mp_obj_exception_clear_traceback(MP_OBJ_FROM_PTR(&MP_STATE_VM(mp_kbd_exception)));
        MP_STATE_VM(mp_pending_exception) = MP_OBJ_FROM_PTR(&MP_STATE_VM(mp_kbd_exception));
        #endif
    }
}

```

[ports/unix/main.c](http://elixir.tronnes.org/circuitpython/latest/source/ports/unix/main.c):
```c
STATIC int execute_from_lexer(int source_kind, const void *source, mp_parse_input_kind_t input_kind, bool is_repl) {
    mp_hal_set_interrupt_char(CHAR_CTRL_C);

    nlr_buf_t nlr;
    if (nlr_push(&nlr) == 0) {
...
        mp_parse_tree_t parse_tree = mp_parse(lex, input_kind);
...
        mp_obj_t module_fun = mp_compile(&parse_tree, source_name, emit_opt, is_repl);

        if (!compile_only) {
            // execute it
            mp_call_function_0(module_fun);
            // check for pending exception
            if (MP_STATE_VM(mp_pending_exception) != MP_OBJ_NULL) {
                mp_obj_t obj = MP_STATE_VM(mp_pending_exception);
                MP_STATE_VM(mp_pending_exception) = MP_OBJ_NULL;
                nlr_raise(obj);
[... 128 lines cut ...]
     usb_msc_background();
     usb_cdc_background();
+
+    if (MP_STATE_VM(mp_pending_exception) != MP_OBJ_NULL) {
+        mp_obj_t obj = MP_STATE_VM(mp_pending_exception);
+        MP_STATE_VM(mp_pending_exception) = MP_OBJ_NULL;
+        nlr_raise(obj);
+    }
+
     last_finished_tick = ticks_ms;
 }

```

It works, but the traceback is gone:
```
Adafruit CircuitPython 4.0.0-alpha-950-g1a6f6ff8e-dirty on 2018-08-18; Adafruit Feather M0 Express with samd21g18
>>>
>>> import time
>>> time.sleep(10)
KeyboardInterrupt:
>>> time.sleep(10)
KeyboardInterrupt:
>>>
```

tannewt, 2018-08-20:
I'm not concerned about a bit of performance loss because its not a core focus for us. I'm also not worried about losing the traceback since its arbitrary. So, its up to you how you want to solve it. :-) Either way works for me.

PR #1142 (2018-08-25): Revert "py/vm: Improve performance of opcode dispatch when using switch stmt."

This reverts commit 869024dd6e62905b7e1069b547856a769b3b24ba.

Ctrl-C stopped producing KeyboardInterrupt with this change on CircuitPython.

The Unix and stm32 ports handles Ctrl-C differently with a handler which is
probably why they where not affected.

Fixes #1092
commit 2d1fecc8 (2018-08-25): Revert "py/vm: Improve performance of opcode dispatch when using switch stmt."

This reverts commit 869024dd6e62905b7e1069b547856a769b3b24ba.

Ctrl-C stopped producing KeyboardInterrupt with this change on CircuitPython.

The Unix and stm32 ports handles Ctrl-C differently with a handler which is
probably why they where not affected.

Fixes #1092
closed 2018-08-25 by dhalbert