adafruit/circuitpython #1093: need driver for TCA9548A I2C Multiplexer

caternuson, 2018-08-06:
This thing:
https://www.adafruit.com/product/2717

ladyada, 2018-08-06:
we dont have a driver for this in arduino - it isnt really drivery cause you only send two bytes. but we could driverify if you think its necessary

deshipu, 2018-08-06:
I think it would make sense to wrap the original i2c object into a class that will automatically switch between the multiplexed buses when you try to use it (when you acquire the lock). This way it would be practically transparent to all other drivers.

ladyada, 2018-08-06:
ooh yeah good point!

caternuson, 2018-08-06:
How would you setup the association of the connected devices with their channels?

ladyada, 2018-08-06:
you'd have it generate a special I2C device-like object that would sneak in the commands for ya

caternuson, 2018-08-06:
The drivers all take in an I2C object and then use `adafruit_bus_device.i2c_device` internally. So we can't pass in a different I2CDevice. I think I see what @deshipu is saying - derive from I2C and override `try_lock()` to sneak in the channel switching commands. That would get called each time the driver uses a context manager for I2CDevice, here:
 https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/master/adafruit_bus_device/i2c_device.py#L105

tannewt, 2018-08-07:
I'd have the TCA9548A take in a `busio.I2C`-like object and have a channel list of `busio.I2C`-like objects. That way I2CDevice will never know the difference.

caternuson, 2018-08-07:
Something more like this?
```python
import board, busio
import some_i2c_sensor
import adafruit_tca9548A

i2c = busio.I2C(board.SCL, board.SDA)
tca = adafruit_tca9548A.TCA9548A(i2c)

# still not sure of best way to provide per channel access
# sensor1 = some_i2c_sensor.Sensor(tca[0])
# sensor1 = some_i2c_sensor.Sensor(tca.chan[0])
# or...?
sensor1 = some_i2c_sensor.Sensor(tca.chan0)
sensor2 = some_i2c_sensor.Sensor(tca.chan1)
sensor3 = some_i2c_sensor.Sensor(tca.chan2)
```

tannewt, 2018-08-08:
Closer! You can have an internal class with `__getitem__` that creates the channel objects on demand and caches them. It'd be `sensor1 = some_i2c_sensor.Sensor(tca.channel[0])`.

caternuson, 2018-08-13:
What should happen if someone tries to re-use the same channel? Like this:
```python
sensor1 = some_i2c_sensor.Sensor(tca.chan[0])
sensor2 = some_i2c_sensor.Sensor(tca.chan[0])
```
Maybe throw a RunTime error that says "Channel in use."?

caternuson, 2018-08-14:
Let's move this discussion to here:
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A/pull/1

caternuson, 2018-08-28:
Closing this for now. See here for driver:
https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A

Thanks @deshipu for the excellent idea on this one - to make the switching happen behind the scenes automagically. Thanks to @tannewt and @ladyada for comments on how best to implement.

closed 2018-08-28 by caternuson