adafruit/circuitpython #1094: String internationalization for Spanish.

C47D, 2018-08-07:
Hi,

My name is Carlos and i will be working on the internationalization of strings for the Spanish language, this is my first contribution so bear with me, any help is welcome!

ladyada, 2018-08-07:
welcome! :)

tannewt, 2018-08-13:
Great @arofarn! Please file a separate issue for French. Let me know if you have any questions. Thanks!

yeyeto2788, 2018-11-24:
@c47d is there anything I can help with the translation into Spanish? As far as I saw in the file there are some string pending to translate. 

Let me know and we can work together on it.

Regards.

C47D, 2018-11-25:
Hi @yeyeto2788, i had been busy with work and as you already noticed there are some not translated or not updated strings, so feel free to send a pull request and me or @sabas1080 can review it.

Gracias por la ayuda!

yeyeto2788, 2018-11-29:
Sorry guys for the late reply, I'll fork the repo to make the changes and then I'll ask for the PR.

Regards.

yeyeto2788, 2018-11-29:
Hello guys! happy to announce that I have done the first PR #1364 I hope it is ok.

There are some things that I wondered while I was translating, I'll list them down here to see what you think.

* There are some typos in english, should I fix them also?

* If I fix a typo in English I guess that it has to be also fixed on the code in order to match the `msgid`, right?

* Regarding words like (`int`, `tuple`, `list`, etc) for me it is ok to have something like this `int (entero)`. 
  Notice that I keep the English word and add the translation for it. What should I do? Right now on the PR you'll see that I have done it for other words but not for the ones listes above.

Please do not hesitate on letting me know what are your thoughts on this.

Regards.

dhalbert, 2018-11-29:
Sure, please fix any English typos as well, and yes, you'll need to find the original location and fix that.

Adding things like `int (entero)` is probably not necessary, since a user has to know the original vocabulary in order to use the language. When in doubt, it's good to be more concise than verbose, since we have limited flash space, especially on M0 boards.

yeyeto2788, 2018-11-29:
@dhalbert thanks for the info. 

I'll check again for those typos and keep in mind about flash space.

closed 2020-07-14 by tannewt