adafruit/circuitpython #1095: PyPI support for Adafruit_CircuitPython_SHT31D

SpitfireMk, 2018-08-07:
First of all, my apologies if I'm not supposed to post this here, but I understand that https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/i2c-sensors-and-devices is suggesting me to open an issue if a library isn't up on PyPI.

So I'm trying to connect a SHT31-D temperature and humidity sensor to a Raspberry Pi 3 model B+ and I can't find the Adafruit_CircuitPython_SHT31D library https://github.com/adafruit/Adafruit_CircuitPython_SHT31D on PyPI.

I would be grateful if you could fix that or tell me if it won't be uploaded in the near future.

Thank you.

kattni, 2018-08-07:
We're working on getting all the libraries into PyPi right now. This one
will be added soon.

kattni, 2018-08-07:
@SpitfireMk The guides and documentation haven't been updated yet, but SHT31D has been added to PyPi and should be ready for you to use.

closed 2018-08-07 by kattni