adafruit/circuitpython #1098: Help translate CircuitPython!

tannewt, 2018-08-08:
Hi!! Thank you for wanting to help translate CircuitPython! We're hoping to translate the supervisor messages like `Press any key to enter the REPL. Use CTRL-D to reload.`, and error messages like `Microphone startup delay must be in range 0.0 to 1.0`. We're not translating any Python keywords.

First, please check to see if there is an existing translation for your language exists [here](https://github.com/adafruit/circuitpython/tree/master/locale).

If the file doesn't exist, then check [the issues list](https://github.com/adafruit/circuitpython/issues?q=is%3Aissue+is%3Aopen+label%3Ainternationalization) to find out if anyone else has begun work and to coordinate with them.

If no issue exists then please create one and reply to this issue with a link. To start a new translation copy [locale/circuitpython.pot](https://github.com/adafruit/circuitpython/blob/master/locale/circuitpython.pot) to locale/<language code>.po where <language code> is an [IETF language tag](https://en.wikipedia.org/wiki/IETF_language_tag) with underscores between subtags.

When submitting feel free to use [GitHub's file editing](https://help.github.com/articles/editing-files-in-another-user-s-repository/) or file upload. Our continuous testing (with Travis) will validate everything before changes are merged. Once they are merged, [test builds are available on S3](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/) under the commit hash shown in [master's commit history](https://github.com/adafruit/circuitpython/commits/master).

For instructions on how to build CircuitPython for testing check out [this guide](https://learn.adafruit.com/building-circuitpython)

turbinenreiter, 2018-08-14:
https://github.com/adafruit/circuitpython/issues/1112

dglaude, 2018-08-15:
Newsletter say "  @arofarn, volunteered to do French " but there is not file in the Github.

sommersoft, 2018-08-23:
While trying to get Travis builds to pass on added module functionality, I'm finding flash space issues with translated builds (non-judgemental! more beautiful languages require more space). In thinking of things to help reduce the hit some, I thought about going through each of the `msgid`s and seeing if we can standardize any of the "close but not exact" iterations. 

Having said that, I'm not sure that it will help that much overall, as I still am not quite sure how it's all getting compiled/linked.

Thoughts?

tannewt, 2018-08-23:
@sommersoft Lets discuss on the feature thread. We've had many space limited builds already and can discuss strategies on a feature by feature basis.

tannewt, 2018-08-23:
Yup, we can combine msgids by changing the `translate` call as we find them.

arofarn, 2018-08-24:
I have seen some duplicated or almost equal strings like "Invalid pin" / "Invalid pins", "unknown type"

About the french translation, I think we can find shorter strings for some of them but it will never be as short as english.

furlaneto, 2018-10-02:
String internationalization for Brazilian Portuguese #1224

yeyeto2788, 2018-12-07:
Hello guys! 

Sorry if this a spam but I'm happy to show some translations in Spanish for CircuitPython (There are some some changes needed but I'll be addressing them soon)

[image]
[image]
[image]
[image]
[image]

Hope you like it.

P.D: there is also a video which I cannot attach to this comment :cry: so here you have link on [Youtube](https://youtu.be/rzQr1jP7TFY)

yeyeto2788, 2018-12-09:
@C47D, estoy de acuerdo. Por eso era importante para mi el hecho de como traducir algunas cosas. Si has pensado en algo no dudes en decírmelo ya que esta semana (Si el tiempo me lo permite) me pondré a traducir el restante.

Saludos.

C47D, 2018-12-09:
@yeyeto2788 Creo que esa era la única convención a la que llegamos, por lo que he visto hiciste muy buen trabajo!. Por mi parte no creo poder revisar las traducciones hasta el siguiente fin de semana.

Saludos

yeyeto2788, 2018-12-10:
@C47D sorry for switching back to English but I also want @tannewt to get the information. I saw some things I wanted to ask.

Is there a way to confirm all string in English are in all translated files? My question is because I thought i have translated everything :cry: but it seems it is not.

I'll try to check the differences and let you guys know but if you know the answer first let me know. 

Regards.

tannewt, 2018-12-11:
@yeyeto2788 The compiler ensures that all built-in exception messages are translatable. There may be others that haven't been set up to be. What are some examples? We can help get everything translatable.

We still may be adding strings as well while we add new features. I hope to get us to a "string freeze" phase before we do the stable release of 4.0.0.

We also do not have a way to provide translations for the strings in our libraries. I would like to add that at some point as well.

yeyeto2788, 2018-12-11:
@tannewt it is just a question I got on my mind since for example, when you do a soft reset of the board the information about CircuitPython version, compilation date and microcontroller is not in Spanish. As far as I saw, that string was translated into Spanish but it is still showing up in English :cry: 

I was also curious about how those errors/strings are added into the files that need to be translated.

I suppose that the libraries might also need to be translated into other languages and IMHO it would require to have the `locale` folder on the libraries itself and then the `upip` can handle the installation of the module. The problem I see in this solution is that either the code gets bigger or we'll need to create/duplicate the modules on each language which is not so good to do on more that 1k modules :worried: 

I can help on translating step by step what is required on my spare time so anything I can help with just ping me.

Regards.

tannewt, 2018-12-13:
@yeyeto2788 Please file separate issues for strings that cannot be translated. That is the best way for us to keep track of them.

The translation files are updated by running `make translate` at the top level of circuitpython. It uses `gettext` to find strings wrapped in a `translate()` function. If I remember right, the version string doesn't work because it uses C preprocessing to create the string rather than printf.

Next year we can take a look at libraries. The bundle can preprocess things as need be for running on CircuitPython and we can do the normal CPython way for it and pip.

deshipu, 2019-03-24:
I started a translation for Polish: https://github.com/adafruit/circuitpython/issues/1690

tannewt, 2019-03-27:
Hi all,
We're nearing completion of the first 4.0.0 stable release. We'll get many more people trying it as a result. Please take a look at translations when you have a chance and make any changes you'd like to see in 4.0.0. We can always update them after as well.
Thanks!

C47D, 2019-03-27:
I will try to get the Spanish translations up to day, haven't looked into it in a while.

wtuemura, 2025-03-31:
Due to health concerns, I regret to inform you that I must step down from my role as the current pt_BR translator for CircuitPython.

I want to sincerely thank you for the opportunity to contribute to this wonderful project.

Wishing the team continued success with CircuitPython!

dhalbert, 2025-03-31:
@wtuemura I am sorry to hear that, and hope your health issues are temporary. Thank you so much for all your work. You have been one of our most reliable and quick translators. We very much appreciate the time you have spent on this translation!