adafruit/circuitpython #1102: PyPi-ification follow-up

kattni, 2018-08-09:
Any left-over issues can be verified in the PyPi Progress spreadsheet.

There are a couple of repos left with badge issues. 
 - LSM9DS1 shows a build error, but the build is passing.
 - CharLCD shows docs failing, they are, but they haven't triggered a new build on the new release which would fix the documentation issue.

Still needing to be added to PyPi:
 - FeatherWing helper lib. The structure of the library caused a number of issues that ended with Sphinx requiring an `autodock_mock_import` of `adafruit_featherwing` which meant that  `api.rst` was no longer populated by the modules, which caused a build failure. Want to revisit this as it should still be added to PyPi.

The plan is to circle back to these issues and sort them at a later date. Any further follow-up that arises can be added to this issue.

kattni, 2018-12-21:
These issues are resolved. The project is ongoing and will continue, but this issue can be closed.

closed 2018-12-21 by kattni

ladyada, 2018-12-21:
nice work @kattni @brennen and all the circuitpython librarians for making this happen. we'll be checking adabot for any missed libraries in the future