adafruit/circuitpython #1103: ITSYM4 Express doesnt respond after downloading code on Ubuntu Linux

labishrestha, 2018-08-09:
Boot_out.txt contents:
Adafruit CircuitPython 3.0.0 on 2018-07-09; Adafruit ItsyBitsy M4 Express with samd51g19

Steps to replicate issue:
1) Plug in ITSYM4 Express to USB port, green led animates, the Neopixel turns on all colors
2) Open mu-editor, load existing code -> Error: **The file "code.py" is already open.**
2) Paste D13 LED blink demo code
3) Click 'Check' -> **Zero problems found**
4) Click 'Save' -> Neopixel Blinks one time, then nothing happens

labishrestha, 2018-08-09:
Additionally, when I click "Serial" Button first time, Mu says : Cannot connect to device on port /dev/ttyACM0
Then if I click 'Serial' button again, (whether or not I hit the reset button and wait), the serial console shows up, but If I press any key, I dont get the >>> prompt and nothing else happens

tannewt, 2018-08-09:
Do you have any other computers you can try it on? Thanks!

labishrestha, 2018-08-13:
I dont have another computer at the moment, but I was able to replicate a possibly related issue on another brand new ITSYM4 board. Here's what I did:

1) Plugged In new board
2) Paste D13 LED blink demo code -> Works fine
3) Forced error in the code by changing the line 'led.value = False' to 'D14.value = False' 
4) Neopixel turned on all colors
5) Reverted the changes made in step 3) -> Neopixel Still full color on, Red LED doesnt blink
6) Reset the board by pressing the reset button -> Neopixel keeps animating green
7) Board seems un-usable at this point.

labishrestha, 2018-08-13:
Update: Was able to test this on a windows 10 machine, and I got both the boards working again after I reflashed CircuitPython

tannewt, 2018-08-14:
Thanks for the reproduction steps! I'll try and look in the next week or so.

tannewt, 2019-03-20:
I haven't been able to reproduce this. @labishrestha please comment if you are able to cause this problem with 4.0.0 Beta 5 or newer. An Itsy Bitsy M4 I have here has 3.0.0-beta.0 on it which is quite old.

closed 2019-03-20 by tannewt