adafruit/circuitpython #1108: Read the Docs seriously out of date

kattni, 2018-08-11:
In adding RTD links to all the guides, I'm finding that some docs are seriously out of date. The issue is that they haven't built in a long period of time. This went unnoticed because when they last built they passed, so the badge has shown passing the entire time.

They do not have webhooks - RTD is instead setup as a service. This needs to be resolved. Only the maintainers on a particular project have access to setup the webhook - adabot and the person who imported it into RTD in the first place.

- [ ] IRRemote
- [ ] MAX31855

sommersoft, 2019-02-17:
@kattni this _appears_ to be good now? All of the listed repos have RTD builds within the last month. Granted, I didn't check any RTDs for libraries not listed here...

kattni, 2019-02-17:
Yeah I think this is probably fine. If anything comes up again, we'll deal with it then.

closed 2019-02-17 by kattni