adafruit/circuitpython #1109: Make it easier to grab a timer and do stuff in background in C modules

deshipu, 2018-08-13:
I have recently been working on a C module that drives a LED matrix directly with GPIO pins, and for that I needed the scanning of the matrix to happen in the background. Normally I would use the system tick timer for this, like I did for the `gamepad` module, but here I needed a little bit higher frequency to prevent flickering.

I found the relevant code needed for this in two places, one is https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/pulseio/PulseOut.c#L100-L138 and the other is https://github.com/adafruit/samd-peripherals/blob/1140ff6d7ed413aa1e2f34c5d847dcc41c924bb9/samd/timers.c#L93

In order to make my module use a timer, I needed to basically copy the code from the `PulseOut.c` — I suppose we could move that into a separate function, to make it a bit easier. I can submit a PR with that if it makes sense, but I'm not sure where it would be best to put that function?

I also had to add a call to my own interrupt handled to `timers.c`, which I consider a bit of a dirty hack, because that file lives in a separate repository and therefore it's impossible to make a change on a separate branch in a single patch.

I'm wondering about a cleaner way of doing this that wouldn't affect efficiency. Just moving that function to the main repository would help a lot already — then we could add our own functions, possibly inside `#ifdef` blocks so that they would only be called when the given module is enabled.

tannewt, 2018-08-14:
I'd suggest moving shared_timer_handler back into CircuitPython and externing it from peripherals.

The code to find a TC could go into peripherals though.

sommersoft, 2018-08-19:
@deshipu,

@tannewt's suggestion to move `shared_timer_handler` back into CircuitPython is underway as part of `samd-peripherals PR #4` referenced above. 

Also at tannewt's suggestion, vectoring which specific handler to use now mimics how it is done with the EIC_Handler (`samd-peripherals/external_interrupts.x`). If you want to plan ahead, and or comment, the current version is [here](https://github.com/sommersoft/circuitpython/commit/3a3c4d80dcd5573881374e91b08d7a8060c2dc8c). This will be part of a PR once I get the `samd-peripherals` PR merged.

deshipu, 2018-08-19:
Oh, thanks, I was planning to work on this this weekend, but things got in a way.

PR #1586 (2019-02-22): Move shared_timer_handler
Fixes #1109.

Part 2, which includes the updated `samd-peripherals`, adds `timer_handler.c/.h`, and updates `common-hal` usage.

Metro M0 Express French was the only failure on my Travis run; hoping I'm missing something and it passes here. 😬
closed 2019-02-22 by tannewt