adafruit/circuitpython #1110: A LED matrix driving module

deshipu, 2018-08-13:
I'm currently experimenting with driving LED matrices (both charlieplexed and normal) directly with the GPIO pins. I have some hacky code that does the deed in the background by using a timer, so all the user needs to do is tell it what pixel should be what shade.

The code that I have so far is at https://github.com/pewpew-game/circuitpython/tree/pewpew70

I'm wondering if there would be an interest in this kind of thing to be merged into the main CircuitPython repository, and if so, what would be a good name for it. Of course it would require a lot of cleanup and some small reorganization.

I plan to have a class for normal rows/columns matrix and a charlieplexed matrix of arbitrary sizes, and I'm also considering adding support for a keyboard matrix, since the code would be very similar anyways.

tannewt, 2018-08-14:
I'm not sure we should have it on by default but I'm happy to have it in the repo. I'd probably call it `matrix` or something similar.

deshipu, 2019-03-24:
This is merged as ``_pew`` already.

closed 2019-03-24 by deshipu