adafruit/circuitpython #1111: Invalid Pins with multiple UART / SPI

robomike, 2018-08-14:
to reproduce.

Run bug.py on an adalogger M0

[smart_robot.zip](https://github.com/adafruit/circuitpython/files/2287369/smart_robot.zip)

when sd mount is last on import i get this:

>>> import bug
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "bug.py", line 4, in <module>
  File "sdmount.py", line 9, in <module>
ValueError: Invalid pins
>>>

You can change the order of import in bug.py and get other invalid pins for UART.

robomike, 2018-08-14:
here is a work around.  Put them in the same file and deinit() them before the other ones.  Importing the modules don't work with this bug.
[bug2.py.txt](https://github.com/adafruit/circuitpython/files/2288484/bug2.py.txt)

robomike, 2018-08-14:
actually the work around is really impractical because it's not modular and it still doesn't work

robomike, 2018-08-14:
i think this is related to https://github.com/adafruit/circuitpython/issues/699, because changing my remote pins to A3 , and A4 fixed the problem

tannewt, 2018-08-15:
Yes, the init order matters because of the SERCOM selection under the hood. What is the bug here? Can this issue be closed?

robomike, 2018-08-19:
yes, this may be closed.

closed 2018-08-19 by robomike