adafruit/circuitpython #1112: String internationalization for German.

turbinenreiter, 2018-08-14:
I'm gonna get started on German translations.

turbinenreiter, 2018-08-14:
https://github.com/adafruit/circuitpython/pull/1114

dglaude, 2018-08-15:
@turbinenreiter , I believe the first step is to create your de.po file and push it here: https://github.com/adafruit/circuitpython/tree/master/locale

This is because the first step for someone wanting to help is to check if the file exist, not to dig in every linked issue on #1098 ...

Tasm-Devil, 2019-01-23:
There is still a lot work to be done. I've got the problem, that I feel uncomfortable to translate native python error messages because it will be harder to search for those errors in the Internet after translations.

Tasm-Devil, 2019-02-03:
houston we have a problem.
The german translations aren't displayed properly in the REPL. The encoding is wrong. 
`ZurÃ¼ckverfolgung (jÃ¼ngste Aufforderung zuletzt)` should be `Zurückverfolgung (jüngste Aufforderung zuletzt)`

closed 2020-07-14 by tannewt