adafruit/circuitpython #1113: String internationalization : french

arofarn, 2018-08-14:
I started to worked on the french translation.

Any help is welcome !

dglaude, 2018-08-15:
I would like to help... but actually since I learned IT in English it is very hard to find the proper french word. I am more of a "franglais" when talking about computer things in French. :-(

I quick and dirty tried to find words frequency to see if there were computer terminology that was "difficult" and translator should agree on. They need to share the same translation for the same word for consistency.

So here is what I can share (can be useful for all languages) and to be discussed for french:

> argument(s) 22+17 paramètre(s)
> objet(s)  27+1 object
> must  47  doit
> invalid 21  invalide
> format  21  format
> type  17 type
> function  16  fonction
> [un]supported 9+14  [non-]supporté
> length  12 longueur
> bytes 12	octets
> [out of]  range	12  [hors des valeurs valides]
> buffer	11	???tampon
> keyword	10	mot-clef
> convert	10  convertir
> file	10	fichier
> requires	9	nécessite
> string	9	chaine de caractère
> value	9	valeur
> 
> [conversion|format] specifier	8	???
> 
> clock 7 horloge
> 
> pins	4	pinoche(???)

dglaude, 2018-08-15:
See #1116

arofarn, 2018-08-15:
I just submitted a PR : https://github.com/adafruit/circuitpython/pull/1116

About some of the words:
arguments => "arguments" or "paramètres" both exist in french
must => "doit" or "nécessite"
bytes => "octets" only if it means the unit (not the type)
out of range => "hors gamme"
keyword => "mot-clef" but keyword-parameter => "paramètre nommé"
pin => I hesitated between "Entrée/sortie", "broche" and just "pin". I finally choose "broche"

For some translations, I looked the french terms in the book "Python 3 Les fondamenteux du langage 2e éd.".

dhalbert, 2018-08-15:
"broche" seems typical, see for instance: https://fr.wikipedia.org/wiki/Atmel_AVR