adafruit/circuitpython #1117: Feather M4 Express and UART problems

arofarn, 2018-08-15:
I need to use 2 UART for a project. The first one with Ultimate GPS featherwing, the second one to communicate with a Raspberry pi. 

It seems that not every couple of pins can handle UART but it does not always trigger Exceptions when I used invalid pins.

Tested with these simples scripts :
- M4 express as sender :

```
import busio, board, time

uart = busio.UART(board.TX, board.RX, baudrate = 9600, timeout=2000, parity=None)

while True:
    t = time.monotonic()
    uart.write("{}\n".format(t))
    print(t)
    time.sleep(1)
```

- M0 express as receiver (as before)
```
 import busio, board, time

uart = busio.UART(board.TX, board.RX, baudrate = 9600, timeout=2000, parity=None)

while True:
    l = uart.readline()
    print(l)
    time.sleep(0.1)
```
I always use TX/RX on the M0 express board.

With TX/RX on M4 express (without Ultimate GPS featherwing) => OK
With D10/D11 or D6/D9 on M4 express (without GPS) => random None or sequences of b'\x00...' on M0 and no error on M4
With D12/D13 or D6/D9 on M4 express (without GPS) => it works

tannewt, 2018-08-15:
I'm confused. Where is the second UART initialized?

arofarn, 2018-08-15:
Whoops, with these scripts I search for pins usable as second UART.
I use 2 UART on my projet : the first for Ultimate GPS featherwing and the second to communicate with a raspberry pi 
Git repository with full code there : https://framagit.org/arofarn/Cameteo/tree/master/circuitpython/code 

Sorry, I first talk about this on the forum and forget to copy the full context :
https://forums.adafruit.com/viewtopic.php?f=57&t=139541

dhalbert, 2018-10-03:
So, summarizing your results above, it appears to you that D10/D11 and D6/D9 on M0 used as UART TX/RX do not work? Everything else appears OK.

ladyada, 2019-08-23:
@tannewt  i think you fixed this issue in 
https://github.com/adafruit/circuitpython/pull/1989
@arofarn - please  try again if you can, will close this issue from lack of activity

closed 2019-08-23 by ladyada