adafruit/circuitpython #1120: Extra \r in Traceback lines

notro, 2018-08-16:
My board to host exception unpickling broke when I pulled the latest master.
There's an extra ```\r``` in every line except the one that prints the exception.
```
cpboard.CPboardRemoteError: b'Traceback (most recent call last):\r\r\n  File "<stdin>", line 21, in <module>\r\r\n  File "<stdin>", line 18, in slave_func\r\r\n  File "<stdin>", line 10, in slave_func\r\r\nKeyboardInterrupt: \r\n'
```

Reverting the translate change fixed it for me:
```diff
diff --git a/py/obj.c b/py/obj.c
index 72bf9bda1..ae0aafd18 100644
--- a/py/obj.c
+++ b/py/obj.c
@@ -86,19 +86,19 @@ void mp_obj_print_exception(const mp_print_t *print, mp_obj_t exc) {
         mp_obj_exception_get_traceback(exc, &n, &values);
         if (n > 0) {
             assert(n % 3 == 0);
-            mp_print_str(print, translate("Traceback (most recent call last):\n"));
+            mp_print_str(print, "Traceback (most recent call last):\n");
             for (int i = n - 3; i >= 0; i -= 3) {
 #if MICROPY_ENABLE_SOURCE_LINE
-                mp_printf(print, translate("  File \"%q\", line %d"), values[i], (int)values[i + 1]);
+                mp_printf(print, "  File \"%q\", line %d", values[i], (int)values[i + 1]);
 #else
-                mp_printf(print, translate("  File \"%q\""), values[i]);
+                mp_printf(print, "  File \"%q\"", values[i]);
 #endif
                 // the block name can be NULL if it's unknown
                 qstr block = values[i + 2];
                 if (block == MP_QSTR_NULL) {
                     mp_print_str(print, "\n");
                 } else {
-                    mp_printf(print, translate(", in %q\n"), block);
+                    mp_printf(print, ", in %q\n", block);
                 }
             }
         }
```

tannewt, 2018-08-17:
What board and version exactly? I don't see double \r in the translation file for trinket on 5dd420f7

notro, 2018-08-18:
Here's a minimal example:

traceback_extra_r.py
```python
import serial
import time

with serial.Serial('/dev/ttyACM0', baudrate=115200) as ser:
    ser.write(b'\x02')  # Ctrl-B
    time.sleep(0.1)
    print(ser.read(ser.inWaiting()))

    cmd = b'raise OSError()\r\n'
    ser.write(cmd)
    time.sleep(0.1)
    print(ser.read(ser.inWaiting()))
```

Result
```
pi@cp:~/work/circuitpython $ python3 test/traceback_extra_r.py
b'\r\n\r\nAdafruit CircuitPython 4.0.0-alpha-950-g1a6f6ff8e-dirty on 2018-08-18; Adafruit Feather M0 Express with samd21g18\r\n>>> '
b'raise OSError()\r\nTraceback (most recent call last):\r\r\n  File "<stdin>", line 1, in <module>\r\r\nOSError: \r\n>>> '
```

tannewt, 2018-11-19:
@notro Can you verify this is fixed? I bet https://github.com/adafruit/circuitpython/commit/355abc835ea75715b4dea38604ef64f8c4eeaa0f fixed it.

notro, 2018-11-19:
Indeed it did, thanks.

closed 2018-11-19 by notro