adafruit/circuitpython #1125: uart scope broken.

robomike, 2018-08-19:
`import board
import busio
import digitalio
import time
import struct

txenable = digitalio.DigitalInOut(board.A2)
esp32_uart = busio.UART(board.TX, board.RX, baudrate=115200, timeout=3000)

def initESPTx():
    
    txenable.switch_to_output(drive_mode=digitalio.DriveMode.OPEN_DRAIN)  ##low enables motor
    txenable.direction = digitalio.Direction.OUTPUT
   
    txenable.value = True

def sendData(data):
    txenable.value = True    
    print(data)
    esp32_uart.write(data)
    txenable.value = False`

Fails at line esp32_uart when SendData is called because esp32_uart is out of scope when this code is imported via REPEL.

`import board
import busio
import digitalio
import time
import struct

txenable = digitalio.DigitalInOut(board.A2)

def initESPTx():
    
    txenable.switch_to_output(drive_mode=digitalio.DriveMode.OPEN_DRAIN)  ##low enables motor
    txenable.direction = digitalio.Direction.OUTPUT
   
    txenable.value = True

def sendData(data):
    esp32_uart = busio.UART(board.TX, board.RX, baudrate=115200, timeout=3000)
    txenable.value = True    
    print(data)
    esp32_uart.write(data)
    txenable.value = False
`
This fixes the scope problem but then maybe the uart didn't have enough time to initialize before sending data, because the baud rate is much too slow.  626Hz was measure for char 0x24 

Finally, this doesn't work .... esp32_uart scope is lost after insert of sleep.

`import board
import busio
import digitalio
import time
import struct

txenable = digitalio.DigitalInOut(board.A2)

def initESPTx():
    
    txenable.switch_to_output(drive_mode=digitalio.DriveMode.OPEN_DRAIN)  ##low enables motor
    txenable.direction = digitalio.Direction.OUTPUT
   
    txenable.value = True

def sendData(data):
    esp32_uart = busio.UART(board.TX, board.RX, baudrate=115200, timeout=3000)
    time.sleep(0.5)
    txenable.value = True    
    print(data)
    esp32_uart.write(data)
    txenable.value = False
`
This issue affects all uart on all sercoms.

dhalbert, 2018-08-19:
Hi - Please quote the exact error messages when submitting bug reports. This is not a scope issue -- it's actually probably bug #1056, which has been fixed but we haven't released a fixed version yet.

I've edited your code samples to use triple backquotes. Do something like this when pasting in code:

    ```
    import busio
    ```

robomike, 2018-08-19:
When the esp32_uart is places bottom of the imports, this is the error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "esp32uart.py", line 21, in sendData
NameError: name 'e' is not defined

Additionally, the latest incremental release could not test the fix because it causes the adalogger to safe-mode .

I call it a scope issue because the uart definition has to be "in the scope" of the same function to work.

dhalbert, 2018-08-19:
Hmm, there is no `e` in the source code examples above, so I don't think I'm looking at the code that's failing for you.

#1056 describes a problem with the underlying UART buffer getting moved around, unbeknownst to the code. That causes UART writes to fail.

When you say "latest incremental release", wht do you mean?

robomike, 2018-08-19:
Kind Sir,

The "e" is the beginning letter of esp32_uart.

last incremental release is adafruit-circuitpython-feather_m0_adalogger-en_US-20180817-5dd420f

robomike, 2018-08-19:
Additionally, this issue still persists on the following older build:
adafruit-circuitpython-feather_m0_adalogger-es-20180814-5f08ebd

deshipu, 2018-08-19:
Can you upload the file with the code that is failing for you? You can do it by zip-ing it and then dragging the zip file into the text entry area. I suspect you have some unicode characters in your code that look like normal letters, but are not, or that are invisible, but cause problems.

robomike, 2018-08-19:
of course, 
[esp32uart.zip](https://github.com/adafruit/circuitpython/files/2300427/esp32uart.zip)

robomike, 2018-08-19:
regarding #1056  .....  usart RX works perfectly fine.  usart TX does not seem to be included in the work for #1056 , i maintain that usart TX is broken.  It either sends nothing, or sends garbage.

robomike, 2018-08-19:
using the Mu editor to make new code.
when executing this code on my adalogger_m0 ; my adalogger becomes unresponsive.
 the REPL is as follows:
Adafruit CircuitPython 3700bc8 on 2018-08-11; Adafruit Feather M0 Adalogger with samd21g18
>>> import send_uart_data
>>> send_uart_data.send_data('0x24')

....

It never returns... 
[send_uart_data.zip](https://github.com/adafruit/circuitpython/files/2300469/send_uart_data.zip)

dhalbert, 2018-08-19:
Some issues with CIRCUITPY writing worked out in discord. Closed in favor of #1126

closed 2018-08-19 by dhalbert