adafruit/circuitpython #1128: uart 4.0

ladyada, 2018-08-20:
Here are some breaking-change requests to make uart match CPython (pyserial)
- Change `timeout` arg to seconds rather than milliseconds
- Require bytes only, don't allow strings (e.g. uart.write("hello") should fail, but uart.write(b'hello') is ok)

tannewt, 2018-08-20:
Sounds good to me!

ladyada, 2018-08-25:
another useful thing, `in_waiting` lets you check if there's something to read, nonblocking 
https://pythonhosted.org/pyserial/pyserial_api.html#serial.Serial.in_waiting

ladyada, 2018-08-29:
i think this ^ is also essentially what https://github.com/adafruit/circuitpython/issues/254 is asking for

ladyada, 2018-09-09:
reset_input_buffer() would also be great, flush input buffer, great for getting clean packets :)

dhalbert, 2018-09-13:
Added a to-do list to initial comment. `UART.in_waiting` and `UART.reset_input_buffer()` implemented by #1186.

PaulKierstead, 2018-09-16:
The addition of in_waiting has made non-blocking timed reads much easier; however, I think it would be very valuable to have the write_timeout separate from timeout, as is the case in pySerial. It is very common to want to wait on writes but not on reads, so separate timeouts are handy. Also useful is out_waiting, in this case so you can implement your own buffering.

I've been trying to do a bunch of asynio type stuff with the UART and those two (along with my 0 timeout fix) would be a very big boon. Of course, having the same units for timeouts would be nice too!

At *some* point cts/rts support might be nice, though they could be added non-breaking later.

ladyada, 2018-09-17:
i think we're slowly going to try and add all pyserial supported functions, but ya some are more useful than others :)

ATMakersBill, 2018-09-19:
Will there be a way to get a UART interface "on top of" the console?  That is, will we be able to use these IO methods to talk to the USB Serial port (ala input() / print())?

If so, I'd like to offer to help test/debug this effort (point me at the right person please).  If not, I think I'll add the "usb_bytes_available" module that Scott requested instead of the input() timeout.

Bill

tannewt, 2018-09-19:
@ATMakersBill That's the idea in #231. We'll have UART like interface for the second USB CDC connection.

ATMakersBill, 2018-09-19:
Hmmmm... second CDC?  Will that show up as a second COM port (or tty on Linux)?  If so, will there be a way to differentiate the two?

dhalbert, 2018-09-19:
Yes, it would be a second COM or tty port, and distinguishing them is probably going to be a problem. In Windows 10 it would just be another "USB Serial Device", unless we had the user install drivers, but we've been trying to get away from that given that Windows 10 has innate `usbser.sys` support. On Windows 7 we would need to add the USB PID to our current driver list, so it would be identifiable in Device Manager. Linux would be another `/dev/ttyACM<n>`.

ATMakersBill, 2018-09-19:
Ok, a separate PID would work.  Also, I believe you can set the "Name" field in the descriptor (it's currently just 
`Name                        : USB Serial Device (COM30)`

Both lsusb on linux and PowerShell on Windows can extract that name

dhalbert, 2018-09-19:
That name (in Windows) is set by the (built-in_ driver). I'm not sure giving a name in the USB descriptor will show up on Device Manager.

ATMakersBill, 2018-09-19:
Interesting - I know that when I plug things and look at them in Arduino they have the device name - I assumed that was sent in in the Name field... perhaps there's a lookup table for the PID?

ATMakersBill, 2018-09-19:
Here's what Windows (PowerShell) can see easily about a Serial Port... perhaps we can control something in here: 

```
__GENUS                     : 2
__CLASS                     : Win32_SerialPort
__SUPERCLASS                : CIM_SerialController
__DYNASTY                   : CIM_ManagedSystemElement
__RELPATH                   : Win32_SerialPort.DeviceID="COM30"
__PROPERTY_COUNT            : 47
__DERIVATION                : {CIM_SerialController, CIM_Controller, CIM_LogicalDevice, 
                              CIM_LogicalElement...}
__SERVER                    : MSI
__NAMESPACE                 : root\cimv2
__PATH                      : \\MSI\root\cimv2:Win32_SerialPort.DeviceID="COM30"
Availability                : 2
Binary                      : True
Capabilities                : 
CapabilityDescriptions      : 
Caption                     : USB Serial Device (COM30)
ConfigManagerErrorCode      : 0
ConfigManagerUserConfig     : False
CreationClassName           : Win32_SerialPort
Description                 : USB Serial Device
DeviceID                    : COM30
ErrorCleared                : 
ErrorDescription            : 
InstallDate                 : 
LastErrorCode               : 
MaxBaudRate                 : 115200
MaximumInputBufferSize      : 0
MaximumOutputBufferSize     : 0
MaxNumberControlled         : 
Name                        : USB Serial Device (COM30)
OSAutoDiscovered            : True
PNPDeviceID                 : USB\VID_239A&PID_8019&MI_00\6&17F6693E&0&0000
PowerManagementCapabilities : {1}
PowerManagementSupported    : False
ProtocolSupported           : 
ProviderType                : Modem Device
SettableBaudRate            : True
SettableDataBits            : True
SettableFlowControl         : True
SettableParity              : True
SettableParityCheck         : True
SettableRLSD                : True
SettableStopBits            : True
Status                      : OK
StatusInfo                  : 3
Supports16BitMode           : False
SupportsDTRDSR              : True
SupportsElapsedTimeouts     : True
SupportsIntTimeouts         : True
SupportsParityCheck         : True
SupportsRLSD                : True
SupportsRTSCTS              : False
SupportsSpecialCharacters   : False
SupportsXOnXOff             : False
SupportsXOnXOffSet          : False
SystemCreationClassName     : Win32_ComputerSystem
SystemName                  : MSI
TimeOfLastReset             : 
PSComputerName              : MSI
```

dhalbert, 2018-12-03:
I have started working on finishing this off.

ladyada, 2018-12-03:
yay!

dhalbert, 2018-12-04:
Changes made. There are bugs discovered in existing UART timeout logic that will be addressed in a new issue.

closed 2018-12-04 by dhalbert

Cyb3r-N1nja, 2022-05-05:
Hello I am new to Circuitpython

I just wanted to ask since strings are not allowed how can I send a Ascii string over Uart and the \n\r (basically ENTER over Uart):

My current code looks like this def get_psuState():                                       # Einlesen der psuState Daten
    uart.write("psuState")
    bytes_psuState = uart.read(173)                    # Read psuState over UART
    string_psuState = ''.join([chr(b) for b in bytes_psuState])
    string_psuState_split = string_psuState.split()     # string sepperate after space
    array_psuState = []
    for line in string_psuState_split:
        if ':' in line:
            i = int(line.split(':')[-1])                    # values after ":" save in array
            array_psuState.append(i)

    BatFault = array_psuState[0]                        # array values to global Variables
    Bat12Fault = array_psuState[1]
    Bat24Fault = array_psuState[2]
    MainsFault = array_psuState[3]
    SensorFault = array_psuState[4]
    SelftestFault = array_psuState[5]
    InitialCharge = array_psuState[6]
    LongtimeTestActive = array_psuState[7]
    batteryOvercurrent = array_psuState[8]
    TotalOvercurrent = array_psuState[9]
    LowBattSwitchOFF = array_psuState[10]