adafruit/circuitpython #1129: NRF52 build fails on MACOS

jerryneedell, 2018-08-20:
With a new clone of the current master repo, I get the following error when trying to build the nrf port on my MACOS system. It compiles fine under Linux.

```
/Users/jerryneedell/projects/circuitpython/ports/nrf
[Jerry-desktop-mini-8:circuitpython/ports/nrf] jerryneedell% make BOARD=pca10056 clean
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
rm -rf build-pca10056-s140 
[Jerry-desktop-mini-8:circuitpython/ports/nrf] jerryneedell% make BOARD=pca10056 
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Create build-pca10056-s140/genhdr/pins.h
QSTR updated
python3 ../../py/makeqstrdata.py build-pca10056-s140/genhdr/qstrdefs.preprocessed.h > build-pca10056-s140/genhdr/qstrdefs.enum.h
python3 ../../py/makeqstrdata.py --compression_filename build-pca10056-s140/genhdr/compression.generated.h --translation build-pca10056-s140/genhdr/en_US.mo build-pca10056-s140/genhdr/qstrdefs.preprocessed.h > build-pca10056-s140/genhdr/qstrdefs.generated.h
FREEZE freeze
arm-none-eabi-gcc -DBLUETOOTH_SD=140 -DBLE_API_VERSION=4 -DBLUETOOTH_SD_DEBUG=1 -mthumb -mabi=aapcs -fsingle-precision-constant -Wdouble-promotion -mtune=cortex-m4 -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Idrivers/bluetooth/s140_nrf52_6.0.0/s140_nrf52_6.0.0_API/include -Idrivers/bluetooth/s140_nrf52_6.0.0/s140_nrf52_6.0.0_API/include/nrf52 -I. -I../.. -Ibuild-pca10056-s140 -Ibuild-pca10056-s140/genhdr -I./../../lib/cmsis/inc -I./boards/pca10056 -I./modules/ubluepy -I./modules/ble -I./nrfx -I./nrfx/hal -I./nrfx/mdk -I./nrfx/drivers/include -I../../lib/mp-readline -I./drivers/bluetooth -I./drivers -I../../lib/tinyusb/src -I./usb -Wall -Werror -ansi -std=gnu99 -nostdlib  -DNRF52840_XXAA -DCONFIG_GPIO_AS_PINRESET -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -fno-strict-aliasing -fstack-usage -fno-builtin -fshort-enums -fdata-sections -ffunction-sections -D__START=main -Os -DNDEBUG -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY   -c -o build-pca10056-s140/frozen_mpy.o build-pca10056-s140/frozen_mpy.c
In file included from supervisor/port.c:32:0:
./common-hal/microcontroller/Pin.h:33:23: error: unknown type name 'pin_obj_t'
 #define mcu_pin_obj_t pin_obj_t
                       ^
./common-hal/pulseio/PWMOut.h:36:11: note: in expansion of macro 'mcu_pin_obj_t'
     const mcu_pin_obj_t *pin;
           ^~~~~~~~~~~~~
make: *** [build-pca10056-s140/supervisor/port.o] Error 1
```

tannewt, 2018-08-20:
Works for me:
```
/V/E/c/p/nrf (translate|✔) $ make BOARD=pca10056 clean
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
rm -rf build-pca10056-s140
/V/E/c/p/nrf (translate|✔) $ make BOARD=pca10056
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Create build-pca10056-s140/genhdr/pins.h
QSTR updated
python3 ../../py/makeqstrdata.py build-pca10056-s140/genhdr/qstrdefs.preprocessed.h > build-pca10056-s140/genhdr/qstrdefs.enum.h
python3 ../../py/makeqstrdata.py --compression_filename build-pca10056-s140/genhdr/compression.generated.h --translation build-pca10056-s140/genhdr/en_US.mo build-pca10056-s140/genhdr/qstrdefs.preprocessed.h > build-pca10056-s140/genhdr/qstrdefs.generated.h
FREEZE freeze
arm-none-eabi-gcc -DBLUETOOTH_SD=140 -DBLE_API_VERSION=4 -DBLUETOOTH_SD_DEBUG=1 -mthumb -mabi=aapcs -fsingle-precision-constant -Wdouble-promotion -mtune=cortex-m4 -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Idrivers/bluetooth/s140_nrf52_6.0.0/s140_nrf52_6.0.0_API/include -Idrivers/bluetooth/s140_nrf52_6.0.0/s140_nrf52_6.0.0_API/include/nrf52 -I. -I../.. -Ibuild-pca10056-s140 -Ibuild-pca10056-s140/genhdr -I./../../lib/cmsis/inc -I./boards/pca10056 -I./modules/ubluepy -I./modules/ble -I./nrfx -I./nrfx/hal -I./nrfx/mdk -I./nrfx/drivers/include -I../../lib/mp-readline -I./drivers/bluetooth -I./drivers -I../../lib/tinyusb/src -I./usb -Wall -Werror -ansi -std=gnu99 -nostdlib  -DNRF52840_XXAA -DCONFIG_GPIO_AS_PINRESET -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -fno-strict-aliasing -fstack-usage -fno-builtin -fshort-enums -fdata-sections -ffunction-sections -D__START=main -Os -DNDEBUG -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY   -c -o build-pca10056-s140/frozen_mpy.o build-pca10056-s140/frozen_mpy.c
LINK build-pca10056-s140/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 226076	   1272	  15476	 242824	  3b488	build-pca10056-s140/firmware.elf
arm-none-eabi-objcopy -O binary build-pca10056-s140/firmware.elf build-pca10056-s140/firmware.bin
arm-none-eabi-objcopy -O ihex build-pca10056-s140/firmware.elf build-pca10056-s140/firmware.hex
Create firmware.uf2
../../tools/uf2/utils/uf2conv.py -f 0xADA52840 -c -o "build-pca10056-s140/firmware.uf2" build-pca10056-s140/firmware.hex
Converting to uf2, output size: 455168, start address: 0x26000
Wrote 455168 bytes to build-pca10056-s140/firmware.uf2.
```

jerryneedell, 2018-08-20:
As suspected - it builds fine in a "case-sensitive filesystem" on the Mac.
Should I just close this or do you want to update the guide first?

tannewt, 2018-08-20:
I added a paragraph about it: https://learn.adafruit.com/building-circuitpython/macos

Thanks!

closed 2018-08-20 by tannewt

ladyada, 2018-08-26:
oof i hit this building on windows too

dhalbert, 2018-08-27:
I am redoing the pin files and this will go away as a side effect of my changes (or I will make sure it goes away).