adafruit/circuitpython #1134: conflict between cpx and neopixel

pvanallen, 2018-08-21:
This code produces an error. With release adafruit-circuitpython-circuitplayground_express-3.0.0.uf2

```
import board
import neopixel
from adafruit_circuitplayground.express import cpx
pixels = neopixel.NeoPixel(board.NEOPIXEL, 10, brightness=.2)
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "cpxneo.py", line 4, in <module>
  File "neopixel.py", line 91, in __init__
ValueError: Pin PB23 in use
```

kattni, 2018-08-21:
`neopixel` is initialised in the Circuit Playground Express `cpx` lib, so you can't import both. If you want to use NeoPixels with the `cpx` lib, try using `cpx.pixels`. Otherwise, you'll need to use the individual libraries you wish to use to be able to use `neopixel` separately.

pvanallen, 2018-08-21:
Ah, got it, thanks. Sorry, I was referencing this [code](https://core-electronics.com.au/tutorials/using-the-accelerometer-with-circuitpython-adafruit-circuit-playground-express.html) which I guess is out of date now.

For completeness, working code:

```
import time
from adafruit_circuitplayground.express import cpx
cpx.pixels.fill((127, 0, 0))
cpx.pixels.show()

while True:
    x, y, z = cpx.acceleration
    print(x, y, z)
    time.sleep(0.1)
```

closed 2018-08-21 by pvanallen

pvanallen, 2018-08-21:
Also, this Adafruit example uses the same approach combining cpx and neopixel libs and doesn't work. It would be good to get that updated:

https://learn.adafruit.com/ufo-circuit-playground-express/code-the-ufo-with-circuitpython

kattni, 2018-08-21:
@pvanallen There is a feedback link on the guide page, if you could leave the feedback there regarding the code, that would be greatly appreciated!

pvanallen, 2018-08-21:
Yes, saw that after my post and commented a few minutes ago, referencing this issue.

ghostdog3, 2022-12-06:
Hello,

I just ran into this same issue while trying to leverage the code on this adafruit guide. [https://learn.adafruit.com/circuitplayground-morse-code-flasher-makecode-circuit-python/circuitpython](url)

I submitted some feedback on the website itself already but in case anyone else runs in this issue, here is what I did:

`pixels = cpx.pixels
pixels.fill((0, 0, 0))
pixels.show()
`