adafruit/circuitpython #1136: Unable to use HID keyboard in boot OS (macOS boot screen)

bmosley, 2018-08-22:
I started putting together a project with my Trinket M0 to use for switching boot volumes on macOS. My goal was to have it select a volume by keystrokes. However, it doesn't seem to be working.Working correctly in macOS once booted

Using the same code for HID keyboard:
1. plug trinket m0 into usb on mac
2. reboot system

`  
import time

import board
import digitalio
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode

# The keyboard object!
time.sleep(1)  # Sleep for a bit to avoid a race condition on some systems
keyboard = Keyboard()
keyboard_layout = KeyboardLayoutUS(keyboard)

while True:
    keyboard.press(Keycode.LEFT_ALT)
    time.sleep(40)
    led.value = False
    keyboard.release_all()
`

dhalbert, 2018-08-22:
The HID keyboard currently presented by CircuitPython is _not_ a  boot device keyboard, with its own endpoint(s), which is what the Mac Open Firmware ("BIOS") and regular PC BIOSes expect. It's part of a composite HID device that includes mouse, multimedia keys, and a gamepad. This was to save endpoints for other purposes.

We've had one other person who wants a boot-compatible keyboard and we're thinking about that - we have to count up the available endpoints and what we want for the future (e.g. USB MIDI). Also, in the long run, we are thinking about how to provide user-configurable HID devices.

bmosley, 2018-08-22:
Thanks. I completely understand and figured that this wasn’t currently supported but thought filing a bug was the best way to ask/suggest implementation of the feature. Is there any roadmap for future HID support? And MIDI sounds quite awesome too :)

dhalbert, 2018-08-22:
No specific timeline with dates, but your request is noted and it's a vote!

bmosley, 2018-08-22:
wonderful, thank you!

ATMakersBill, 2020-02-09:
This is marked as closed - was it done?  If the fix was the same that I used in the XBox stuff which requires a custom build, is there any example of making this work?

dhalbert, 2020-02-09:
The fix requires a new descriptor and its own endpoint, at least. I'm not sure what else. Perhaps it could be added as a compile option for the descriptor generating Python script, and then later as a `boot.py`. We can reopen this as long term.

ATMakersBill, 2020-02-09:
I understand this not being in the core CP, but I do think it should remain open until there's some way to do it.  I don't mind having to do a custom build (like we do for the XBox Adaptive Controller), but we have found lots of folks who really want to use devices through KVMs, and the composite devices don't work through them.
Here's our current pressing issue: https://www.facebook.com/groups/ATMakers/permalink/815702142174560/

ATMakersBill, 2020-02-09:
BTW, thanks Dan

dhalbert, 2020-02-09:
Do you need a bootmode mouse HID capability too?

ATMakersBill, 2020-02-09:
Um... I don't think so?  I think the reason this is needed is that the KVM listens in on the keyboard commands and relays them.  That lets it watch for trigger keystrokes ('scroll-scroll-enter') to jump between devices.  We don't kneed the triggers, but the stupid KVM doesn't know what to do with composite devices.  It might be that we just need a non-composite device w/no other profiles.  Or only a single device (no serial/mass storage?)  

I just know it is a known issue that "complicated" keyboards (even those with media keys) don't work.

bitboy85, 2021-03-05:
Here is another vote for it :)
By now i've tried arduino, vusb, teensy, and pico pi and only teensy seems to have a full featured keyboard that works in BIOS and KVM devices.
The teensy itself is also a composite device, having keyboard, mouse, serial and still works fine with a KVM. So my best guess is, that it only depends on a correct implementation of the boot keyboard.
As far as i know, a boot keyboard needs its own descriptor and endpoint and on top of it, it has two additional functions which are described in the usb docs.

dhalbert, 2021-03-05:
We are currently out of endpoints on most boards with 8 endpoints, having added a second CDC. If someone would like to contribute boot keyboard and mouse descriptors and compile flags, we'd be happy to include them. We would not turn them on normally, since they would remove having MIDI or the second CDC available by default.

This is another indicator for user-configurable USB and USB HID descriptors. However, we are pretty much out of space on the smallest builds, and don't have room to add the code for this. It would have to be on something other than the non-Express SAMD21's.

hathach, 2021-03-05:
@dhalbert boot keyboard doesn't need to be on its own interface and/or take extra endpoint. It can be used with existing HID device. Basically the HID device has 2 mode (or protocol): boot mode, and report mode (default). 
- In report mode: it is composite (keyboard + mouse + media key)
- In boot mode: it only report 8 bytes of keyboard report (no report ID)

Most bios will always start with SET_PROTOCOL = BOOT upon enumeration. We only need to handle the callback, set the mode correctly then skip all other report except keyboard. I will update my hid_composite example in my repo so that you could have a clear example to follow in circuitpython later on. Will tag you on the modification later on, may try to submit PR myself as well. Though for the flash size, it may indeed increase a bit of space.

dhalbert, 2021-03-05:
Aha, great, thanks! I will await your example. It would be more complicated to add a boot mouse, but at least having a keyboard will satisfy many needs.

hathach, 2021-03-05:
each interface can only either support boot keyboard or mouse, to support both, we need an extra interface though.

deshipu, 2021-03-05:
But when in boot mode, you don't need MSC and CDC and MIDI, no?

dhalbert, 2021-03-05:
> But when in boot mode, you don't need MSC and CDC and MIDI, no?

Good point - it's more a question of having code space on the smallest builds to switch back and forth.

bitboy85, 2021-03-05:
@hathach Are you sure about the not needed extra endpoint? I tried a few times getting this to work with the arduino environment and to have an idea where the issue is i used this tool https://www.thesycon.de/eng/usb_descriptordumper.shtml and a real keyboard always used an extra endpoint.

@dhalbert
during my research i never found someone asking for a boot mouse. Most BIOSes are configured by keyboard and Hotkeys in KVMs are also done with a keyboard

tannewt, 2021-03-09:
I believe that BOOT mode is missing n-key rollover so we'll want to switch between the two if we can.

hathach, 2021-03-09:
@bitboy85 can you post the descriptor of your real keyboard here along with its hid descriptor for the reference. If it is too long, you could move it to a gist and have a link here.

bitboy85, 2021-03-09:
@hathach
https://gist.github.com/bitboy85/543b34a9505280ead8103a4d4d9cab26
2 different keyboards, no mixed device.

I remember a doc that says, a boot keyboard endpoint requires a packet size of 8 byte.

As additional reference: https://www.usb.org/sites/default/files/hid1_11.pdf
page 50-54: GET_PROTOCOL and SET_PROTOCOL request for boot device
page 62: Keyboard implementation
page 74: Boot keyboard requirements

tannewt, 2021-03-09:
@obra may know what we need to be a boot keyboard too.

akaenner, 2021-03-11:
I also need a keyboard that works at boot time. I'm currently working on a software for a DIY split keyboard with each part having a Raspberry Pi Pico in it. All features I planned are now implemented and I will share the code on GitHub in the comming weeks. I just realized, that it does not work at boot time (at least not under macOS). I hope that this issue can be fixed, because otherwise building fully fledged custom keyboards is out of reach by using CircuitPython.

obra, 2021-03-12:
You want to have a look at the descriptors we use for https://github.com/keyboardio/KeyboardioHID
The idea is that you should offer up both a Boot Protocol descriptor and a bitmap descriptor.

bitboy85, 2021-04-08:
As it is possible to config USB devices on a custom build, any hints where to start when trying to implement this?
https://learn.adafruit.com/building-circuitpython/customizing-usb-devices

I've read some docs about the requirement.
- A boot keyboard only does 6kro. But this is already the case in the current implementation
- The interface descripter field bInterfaceSubClass should be set to 0x01 (Boot_Protocol)
- A boot protocol keyboard must support GET_REPORT, GET_IDLE, SET_IDLE, GET_PROTOCOL and SET_PROTOCOL requests. (https://www.ftdichip.com/Support/Documents/AppNotes/an_345_ft51a_keyboard_sample.pdf p. 12) where is the source file to handle those class requests?
- Some USB-Hosts might have a buggy implementation and do not send SET_PROTOCOL https://www.devever.net/~hl/usbnkro https://deskthority.net/wiki/USB
- I'm sure i read somwhere that the Endpoint packet size should be 8 bytes, but i can't find it anymore.

dhalbert, 2021-04-08:
I am starting to implement dynamic control over USB and USB HID descriptors right now. I may not get to boot keyboard immediately, but it will be possible. So I would say not to start on it yourself right now, because the underlying code is going to change a lot in the near future.

All that we really need is a separate USB descriptor (separate endpoint) with an appropriate USB HID descriptor. Almost any commercial keyboard does this and we can vet our choices against the descriptors from such keyboards.

azcaban, 2021-04-26:
interested in the request. i have a macropad that works with circuitpython, which i would like to use to access to bios. here the thing [https://learn.adafruit.com/itsybitsy-keybow-mechanical-keypad?view=all](url)

dhalbert, 2021-04-26:
I am not implementing boot keyboard in the first pass of dynamic USB descriptors, which I am working on right now. But internally the implementation is flexible enough to add later. Once I get the first version working it can be revisited.

hathach, 2021-08-17:
I am adding the `hid_boot_interface` example here if anyone is interested to test https://github.com/hathach/tinyusb/pull/1025 . I don't have any legacy bios PC to test with though. Therefore it may have an issue or two. For implementation it uses 1 interface for keyboard, and 1 for mouse. Per HID specs, device is default to report mode, Bios PC will send control request SET_PROTOCOL to change it to boot mode which cause this callback to invoked https://github.com/hathach/tinyusb/pull/1025/files#diff-197babc2310c6d123603f9f2df568b1c1f1857cee19037e747408f146a847d4fR170-R176

You could also use `tud_hid_n_get_protocol()` to get the current mode and send construct the report accordingly.
- If `report` mode: report is sent accordingly to descriptor like we are currently doing (can be composite)
- If boot mode: only send 8 bytes of keyboard report (no ID) since we implement boot keyboard.

dhalbert, 2021-10-08:
I am now working on this. I ran into an issue of BIOS's expecting the interface number to be 0 for the keyboard, and not handling non-zero values properly. See https://github.com/hathach/tinyusb/issues/1129. I can push the HID interface to position 0, but right now the Windows drivers package we have assumes the CDC is in position 0. I think I have a way of restructuring the drivers to handle that, though, by imitating [what Windows 10 does](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-driver-installation-based-on-compatible-ids#windows10) for `usbser.sys`. I'll test that it, which might considerably simplify the Windows 10 driver package as well.

dhalbert, 2021-10-11:
I am going to put this on ice for now. As long as we continue to support Windows 7 and 8.1, we cannot move the Device Interface numbers around so that HID is Interface 0 instead of CDC. In the long run, it could be possible, but boot keyboard support may become less important. I have some BIOS examples that don't need boot keyboards anyway.

akaenner, 2021-10-13:
How difficult is it to create a branch of CircuitPython with the HID device at position 0? 

I'm currently writing a documentation for my upcoming DIY keyboard. The firmware is based on CircuitPython and is much easier to use than others like QMK. Not being able to emulate a boot keyboard is of course a big show stopper. For example: under macOS you can not login without a boot keyboard if your hard disk is encrypted and most users encrypt their HD.  

See [image]

akaenner, 2021-10-16:
Hi Dan,

I compiled your hid-boot-protocol branch and tested it with my keyboard but I'm sure I'm doing something wrong. Maybe you can help me. Here is the message in boot_out.txt:

```
Adafruit CircuitPython 5.0.0-alpha.0-11771-g13210d2cc-dirty on 2021-10-15; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico
boot.py output:
Traceback (most recent call last):
  File "boot.py", line 41, in <module>
TypeError: extra keyword arguments given
```

Here is the code in line 41 in boot.py:
`usb_hid.enable((usb_hid.Device.KEYBOARD,), boot_protocol=1 if not maintenance_mode else 0)`

akaenner, 2021-10-16:
ok, here is the part of my boot.py which disables other devices and enables just the keyboard. 
```
if not maintenance_mode:
    storage.disable_usb_drive()
    usb_cdc.disable()
    usb_midi.disable()
    
usb_hid.enable((usb_hid.Device.KEYBOARD,), boot_device=1 if not maintenance_mode else 0)
```

The USB-Stick does not show up in the Finder  an if I'm already logged in to the mac, the keyboard works.
But the keyboard still does not work when I try to login on a machine with an encrypted HD.

Here is the output from lsusb -v (only the pico part)

```
Bus 020 Device 014: ID 239a:80f4  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x239a 
  idProduct          0x80f4 
  bcdDevice            1.00
  iManufacturer           1 Raspberry Pi
  iProduct                2 Pico
  iSerial                 3 E6609CB2D37A3F29
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          195
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 None
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 None
      iInterface              4 CircuitPython CDC control
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x01
          call management
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
[... 161 lines cut ...]
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       3
Device Status:     0x0000
  (Bus Powered)
```

akaenner, 2021-10-17:
I forgot to import usb_cdc. Now the output from lsusb command for the Pico looks better:

```
Bus 020 Device 013: ID 239a:80f4  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x239a 
  idProduct          0x80f4 
  bcdDevice            1.00
  iManufacturer           1 Raspberry Pi
  iProduct                2 Pico
  iSerial                 3 E6609CB2D37A3F29
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              4 CircuitPython HID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               8
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               8
Device Status:     0x0000
[... 74 lines cut ...]
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

dhalbert, 2021-10-17:
@akaenner I have a simpler test of this, which is to start the Mac while holding down the "D" key, to get it into diagnostics mode. Then it lists the available wifi networks (wait for the spinning to stop), and an external keyboard's up and down arrow keys can go up and down the list of networks.

The "D" key is for Intel Macs. For M1 ("Apple silicon"), you just keep holding down the power button to get into a startup menu. https://support.apple.com/en-us/HT202731

Anyway, what I see on an Intel MacBook is that a MacroPad set up for boot keyboard and sending up and down arrows does not work, but a vanilla Dell external keyboard does. So this gives me something to work toward.

akaenner, 2021-11-26:
I just released my keyboard firmware on github:  https://github.com/akaenner/picosplit
And the building instructions for my raspberry pi Pico and CircuitPython based keyboard: https://kaenner.de/picosplit

DietmarSchwertberger, 2021-11-26:
@akaenner : On a quick look I did not see exception handlers. See [PR #1929](https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/1929). Without, on my Windows PCs the software would not re-connect on re-boot or when the BIOS takes a few seconds before Windows is booting. If you have exception handlers, please point me to the positions, as I would like to update the PR with the actual exception types instead of catching all.

akaenner, 2021-11-28:
@DietmarSchwertberger : I added an exception handler which does a reset. Do you think this is enough? I tested it with a PC running Windows and waking up the computer by pressing a key still does not work. Here is my code: https://github.com/akaenner/picosplit/blob/main/Master/main.py

deshipu, 2021-11-28:
You really do not want to use a naked "except" like this, ever.

It will catch KeyboardInterript, MemoryError and even SystemExit exceptions.

Please use `except Exception:` if you really must do pokemon programming, but if possible, find out which exceptions you want to catch and only catch those.

DietmarSchwertberger, 2021-11-28:
@akaenner : a reset seems a bit hard and unnecessary.
I have tried with sending my Windows PC to Standby. The exception will only be triggered when the keyboard is trying to send the keypress. Then, `Keyboard(usb_hid.devices)` will only succeed, once the PC is up and running again. During standby it will constantly fail.
I would guess that the wake-up is done by something outside the normal HID protocol.

bitboy85, 2021-12-05:
I've tried with a Lenovo T410 Notebook. Sadly a keypress from the device completly locks up the BIOS. Even the normal keyboard is not working anymore. This happens at the time where the first keypress is send from the CP-device. Just connecting the CP-device does not trigger this issue.
Every following keypress then gives an "error" sound from the bios, no matter if its from the CP-device or from the notebooks own keyboard. In normal mode this sound is played if you press a key which has no effect in the current state. Like pressing "right" when there is no menu to the right.
A keypress from the CP-device also triggers this sound, so my guess is, the keyboard and the keypress is correctly detected

retrospecced, 2021-12-30:
Thank you @dhalbert and everyone for everything, I couldn’t get CircuitPython’s HID to work with my KVM either, due to the needed boot keyboard protocol. Special shout-out to @obra and the boot keyboard descriptors in https://github.com/keyboardio/KeyboardioHID, I used it to modify (slightly) the boot descriptor dhalbert graciously provided for a related issue https://github.com/hathach/tinyusb/issues/1129#issuecomment-937756019, and now it works with my KVM!  

CP 7.1.0 rc1 on a Feather Express M4 and using this boot.py

<details><summary>boot.py</summary>
<p>

```Python
import board
import digitalio
import storage
import usb_cdc
import usb_midi
import usb_hid
from time import sleep

switch = digitalio.DigitalInOut(board.D5)
switch.direction = digitalio.Direction.INPUT
switch.pull = digitalio.Pull.UP
sleep(0.1)

BOOT_KEYBOARD_DESCRIPTOR=bytes((
    0x05, 0x01,        # Usage Page (Generic Desktop Ctrls)
    0x09, 0x06,        # Usage (Keyboard)
    0xA1, 0x01,        # Collection (Application)
    0x05, 0x07,        #   Usage Page (Kbrd/Keypad)
    0x19, 0xE0,        #   Usage Minimum (0xE0)
    0x29, 0xE7,        #   Usage Maximum (0xE7)
    0x15, 0x00,        #   Logical Minimum (0)
    0x25, 0x01,        #   Logical Maximum (1)
    0x75, 0x01,        #   Report Size (1)
    0x95, 0x08,        #   Report Count (8)
    0x81, 0x02,        #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x01,        #   Report Count (1)
    0x75, 0x08,        #   Report Size (8)
    0x81, 0x01,        #   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x05,        #   Report Count (3)
    0x75, 0x01,        #   Report Size (1)
    0x05, 0x08,        #   Usage Page (LEDs)
    0x19, 0x01,        #   Usage Minimum (Num Lock)
    0x29, 0x05,        #   Usage Maximum (Kana)
    0x91, 0x02,        #   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0x95, 0x01,        #   Report Count (1)
    0x75, 0x03,        #   Report Size (5)
    0x91, 0x01,        #   Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0x95, 0x06,        #   Report Count (6)
    0x75, 0x08,        #   Report Size (8)
    0x15, 0x00,        #   Logical Minimum (0)
    0x26, 0xFF,        #   Logical Maximum (255)
    0x05, 0x07,        #   Usage Page (Kbrd/Keypad)
    0x19, 0x00,        #   Usage Minimum (0x00)
    0x2A, 0xFF,        #   Usage Maximum (0xFF)
    0x81, 0x00,        #   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,
))

boot_keyboard = usb_hid.Device(
    report_descriptor=BOOT_KEYBOARD_DESCRIPTOR,
    usage=0x06,
    usage_page=0x01,
    report_ids=(0,),
    in_report_lengths=(8,),
    out_report_lengths=(1,),
    )

maintenance_mode = not switch.value
if not maintenance_mode:
    storage.disable_usb_drive()
    usb_cdc.disable()
    usb_midi.disable()
    usb_hid.enable((boot_keyboard,), boot_device=1 )
else:
    usb_hid.enable((usb_hid.Device.KEYBOARD,))

```

</p>
</details>

dhalbert, 2021-12-30:
@retrospecced Thanks for reporting this. I compared your descriptor above with the one I referenced, and here are the diffs. Your changes are on the right hand side. See the `|` in the middle which marks the changed lines.
```c
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)	0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)				0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)			0xA1, 0x01,        // Collection (Application)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)		0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)			0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)			0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)			0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)			0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)				0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)			0x95, 0x08,        //   Report Count (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Pr	0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Pr
0x95, 0x01,        //   Report Count (1)			0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)				0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear	0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear
0x95, 0x03,        //   Report Count (3)		      |	0x95, 0x05,        //   Report Count (3)
0x75, 0x01,        //   Report Size (1)				0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)			0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)		0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)			0x29, 0x05,        //   Usage Maximum (Kana)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,P	0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,P
0x95, 0x01,        //   Report Count (1)			0x95, 0x01,        //   Report Count (1)
0x75, 0x05,        //   Report Size (5)			      |	0x75, 0x03,        //   Report Size (5)
0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linea	0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linea
0x95, 0x06,        //   Report Count (6)			0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)				0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)			0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)		      |	0x26, 0xFF,        //   Logical Maximum (255)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)		0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)			0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)		      |	0x2A, 0xFF,        //   Usage Maximum (0xFF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,	0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,
0xC0,              // End Collection				0xC0,              // End Collection
```
In yours, there are 5 bits of padding and 3 bits of LED's (the comments are incorrect, the `3` and `5` should be reversed. Also, the two-byte `0xFF, 0x00` Maximums are replaced by the one-byte `0xFF`. According to http://eleccelerator.com/usbdescreqparser/, yours does not parse properly. Yet it seems to work, so I am not sure why.

`0x26` (`0b100110`) means, for instance, that it is a Logical Maximum, 2 bytes long. The value is signed, so 255 has to be represented as two bytes, otherwise the `0xFF` would be interpreted as `-1`. The lower two bits are the length, so `10` means 2 two bytes. So if that's the descriptor you got from the other repo, it is slightly incorrect.

The actual descriptor does not necessarily matter, because when you set `boot_device=1`, the supplied HID report descriptor is ignored, and a standard one is used instead. See page 59 in https://www.usb.org/sites/default/files/hid1_12.pdf for this standard descriptor.

Nevertheless, yours works, in your situation. If you change the things I mentioned, does it make any difference?

retrospecced, 2021-12-30:
Thank you for the analysis on the descriptor, and your corrections to mine also work with my KVM, as you probably expected!

When I set boot_device=1, the KVM doesn't work unless I supply your descriptor as a parameter.  I don't think it's ignoring it.

```Python
#usb_hid.enable((usb_hid.Device.KEYBOARD,), boot_device=1 )
# above line does not work with Belkin SOHO KVM

#usb_hid.enable((reference_keyboard,), boot_device=1 )
# above line works fine with Belkin SOHO KVM
```

<details><summary>boot.py (updated)</summary>
<p>

```Python
"""CircuitPython Essentials Storage logging boot.py file"""
import board
import digitalio
import storage
import usb_cdc
import usb_midi
import usb_hid
from time import sleep

switch = digitalio.DigitalInOut(board.D5)

switch.direction = digitalio.Direction.INPUT
switch.pull = digitalio.Pull.UP
sleep(0.1)

REFERENCE_BOOT_KEYBOARD_DESCRIPTOR=bytes((
0x05, 0x01,        # Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        # Usage (Keyboard)
0xA1, 0x01,        # Collection (Application)
0x05, 0x07,        #   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        #   Usage Minimum (0xE0)
0x29, 0xE7,        #   Usage Maximum (0xE7)
0x15, 0x00,        #   Logical Minimum (0)
0x25, 0x01,        #   Logical Maximum (1)
0x75, 0x01,        #   Report Size (1)
0x95, 0x08,        #   Report Count (8)
0x81, 0x02,        #   Input (Data,Var,Abs,No Wrap,Linear,Pr
0x95, 0x01,        #   Report Count (1)
0x75, 0x08,        #   Report Size (8)
0x81, 0x01,        #   Input (Const,Array,Abs,No Wrap,Linear
0x95, 0x03,        #   Report Count (3)
0x75, 0x01,        #   Report Size (1)
0x05, 0x08,        #   Usage Page (LEDs)
0x19, 0x01,        #   Usage Minimum (Num Lock)
0x29, 0x05,        #   Usage Maximum (Kana)
0x91, 0x02,        #   Output (Data,Var,Abs,No Wrap,Linear,P
0x95, 0x01,        #   Report Count (1)
0x75, 0x05,        #   Report Size (5)
0x91, 0x01,        #   Output (Const,Array,Abs,No Wrap,Linea
0x95, 0x06,        #   Report Count (6)
0x75, 0x08,        #   Report Size (8)
0x15, 0x00,        #   Logical Minimum (0)
0x26, 0xFF, 0x00,  #   Logical Maximum (255)
0x05, 0x07,        #   Usage Page (Kbrd/Keypad)
0x19, 0x00,        #   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  #   Usage Maximum (0xFF)
0x81, 0x00,        #   Input (Data,Array,Abs,No Wrap,Linear,
0xC0,              # End Collection
))

reference_keyboard = usb_hid.Device(
    report_descriptor=REFERENCE_BOOT_KEYBOARD_DESCRIPTOR,
    usage=0x06,
    usage_page=0x01,
    report_ids=(0,),
    in_report_lengths=(8,),
    out_report_lengths=(1,),
    )
    
maintenance_mode = not switch.value
if not maintenance_mode:
    storage.disable_usb_drive()
    usb_cdc.disable()
    usb_midi.disable()
    usb_hid.enable((reference_keyboard,), boot_device=1 )
else:
    usb_hid.enable((usb_hid.Device.KEYBOARD,))

#usb_hid.enable((usb_hid.Device.KEYBOARD,), boot_device=1 )
# above line does not work with Belkin SOHO KVM

#usb_hid.enable((reference_keyboard,), boot_device=1 )
# above line works fine with Belkin SOHO KVM

"""output from this boot.py
Adafruit CircuitPython 7.1.0-rc.1 on 2021-12-25; Adafruit Feather M4 Express with samd51j19
Board ID:feather_m4_express
boot.py output:
"""
```
</p>
</details>

Again, thanks for your time, let me know if I can test anything else for you, as I am super-thankful of all of your time and effort spent on making CircuitPython work for us (especially for first-time Python programmers like myself!)

dhalbert, 2021-12-30:
> When I set boot_device=1, the KVM doesn't work unless I supply your descriptor as a parameter. I don't think it's ignoring it.

You do need a descriptor. How it works is a bit more complicated than one might expect. If you set `boot_device=1`, then CircuitPython says it is capable of being a boot keyboard. But it doesn't do so unless the host asks for it to be a boot keyboard. If the host doesn't ask, then the supplied descriptor is used. If the host does ask, then the supplied descriptor is sent but ignored, and the host assumes the "standard" one.

bitboy85, 2022-01-02:
Did some further testing and i think this is a weird one. I didn't change the descriptor. I just added usb_hid.get_boot_device() to check if the host requests boot mode -> Yes it does.

So i said it locks up bios or grub boot loader. I found this is only half true.
After sending the first keypress it locks up. Nether from the cp-device nor from the original keyboard an input is accepted.
EXCEPT: pressing and releasing the left CTRL-key does unlock it. The normal keyboard starts working again and even better, the cp-device now acts as a keyboard perfectly fine.

I do not use CTRL key anywhere in my script and i don't have any issues without cp-device so i don't think the error comes from the built-in keyboard.

The behaviour is like a permanently pressed ctrl button. CTRL left button. But only one time after pluggin it in.
Pressing ALT + DEL while in bios reboots the pc.

But i can't find a part in my code where i send ctrl without releasing it.

So its fixable with kbd.release(Keycode.CONTROL) at the beginning of my script. But it doesn't explain why it is pressed in the first place. Even more confusing: https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/main/adafruit_hid/keyboard.py does a release_all in the init part. Maybe something is going wrong there. I'll take a closer look tomorrow.

retrospecced, 2022-01-02:
I'm using a KVM, but what you describe is the same behavior I was seeing before I started using the custom descriptor. https://w3c.github.io/uievents/tools/key-event-viewer.html would report left-control was always pressed before my code ran.   Also, the KVM wouldn't forward any shift key presses to the host.  So an "A" was always showing as control-a (lowercase), and so on.

justinesmithies, 2022-02-27:
> Thank you for the analysis on the descriptor, and your corrections to mine also work with my KVM, as you probably expected!
> 
> When I set boot_device=1, the KVM doesn't work unless I supply your descriptor as a parameter. I don't think it's ignoring it.
> 
> ```python
> #usb_hid.enable((usb_hid.Device.KEYBOARD,), boot_device=1 )
> # above line does not work with Belkin SOHO KVM
> 
> #usb_hid.enable((reference_keyboard,), boot_device=1 )
> # above line works fine with Belkin SOHO KVM
> ```
> 
> boot.py (updated)
> 
> Again, thanks for your time, let me know if I can test anything else for you, as I am super-thankful of all of your time and effort spent on making CircuitPython work for us (especially for first-time Python programmers like myself!)

Just want to thank @retrospecced as using his boot.py resolved my issues whereby I couldn't access my ThinkPads BIOS or enter the grub menu on Arch Linux.
Now I just need to see if it's possible to change my Pico manufacturer info as lsusb shows Adafruit Pico and I'd like to change it to something else.

bitboy85, 2022-02-28:
@retrospecced thx for your effort.
i tried your descriptor (and boot.py) but for some reason the device gets an exclamation mark in the device manager. So i was unable to test with the mentioned startech kvm switch.

dhalbert, 2022-02-28:
@retrospecced's original descriptor has some errors. Did you try the left-hand one in the diff above.

bitboy85, 2022-03-09:
Ah thx. I've thought the provided source does already contain the fix.

Meanwhile i found another descriptor and can confirm working with a startech kvm switch 👍 
Other USB functions like storage, midi, and cdc needs to be turned off.
I'm using Circuitpython 7.2.0

```
BOOT_KEYBOARD_DESCRIPTOR=bytes((
    0x05, 0x01,        # Usage Page (Generic Desktop Ctrls)
    0x09, 0x06,        # Usage (Keyboard)
    0xA1, 0x01,        # Collection (Application)
    0x75, 0x01,        # Report Size (1)
    0x95, 0x08,        # Report Count (8)    
    0x05, 0x07,        # Usage Page (Kbrd/Keypad)
    0x19, 0xE0,        # Usage Minimum (0xE0, 224)
    0x29, 0xE7,        # Usage Maximum (0xE7, 231)
    0x15, 0x00,        # Logical Minimum (0)
    0x25, 0x01,        # Logical Maximum (1)
    0x81, 0x02,        # Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x01,        # Report Count (1)
    0x75, 0x08,        # Report Size (8)
    0x81, 0x03,        # Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x05,        # Report Count (5)
    0x75, 0x01,        # Report Size (1)
    0x05, 0x08,        # Usage Page (LEDs)
    0x19, 0x01,        # Usage Minimum (Num Lock)
    0x29, 0x05,        # Usage Maximum (Kana)
    0x91, 0x02,        # Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0x95, 0x01,        # Report Count (1)
    0x75, 0x03,        # Report Size (3)
    0x91, 0x03,        # Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0x95, 0x06,        # Report Count (6)
    0x75, 0x08,        # Report Size (8)
    0x15, 0x00,        # Logical Minimum (0)
    0x25, 0x68,        # Logical Maximum (104)
    0x05, 0x07,        # Usage Page (Kbrd/Keypad)
    0x19, 0x00,        # Usage Minimum (0)
    0x29, 0x68,        # Usage Maximum (104)
    0x81, 0x00,        # Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,
))
```

RefactorFactory, 2022-07-08:
I have a theory of the various keyboard descriptors discussed and the problem where the **left Ctrl key** seems to be pressed:

If the device sends reports prefixed with a 1-byte **report ID** of `0x01`, the host is not expecting the prefixed report ID byte, and it interprets it as the **modifiers field**, and a modifiers field of `0x01` corresponds to the left Ctrl key. By calling `usb_hid.enable((reference_keyboard,)...)` with `reference_keyboard` having a **report ID of 0**, that causes CircuitPython to pass a report ID of 0 to TinyUSB which **prevents it from prefixing reports with a report ID**, solving the issue. The implication is that perhaps the *contents* of the keyboard descriptor doesn't matter as much as using a **report ID of 0**. **Conclusion: for boot keyboards, always use a report ID of 0 for simple BIOSes**. Maybe something to consider for CircuitPython's defaults, so that users don't need to specify a custom keyboard descriptor.

I may have found a way to **keep CDC and possibly other functions enabled** while also using a boot keyboard:

Keep the other functions enabled, but make the boot keyboard use an **endpoint address of EP 1 IN**. By doing this with plain Arduino (no CircuitPython), I was able to use a boot keyboard and CDC with my old PC.

zan73, 2025-05-27:
Having tried a few variants of the boot descriptor from above comments and also just
usb_hid.enable((Device.KEYBOARD,), boot_device=1)
as documented [here](https://docs.circuitpython.org/en/latest/shared-bindings/usb_hid/#usb_hid.enable), I am still not able to get a working boot mode keyboard implementation to work with MacOS cold-boot (FileVault) login.

Can anyone confirm that they have boot-mode keyboard working for MacOS? (specifically a cold-boot, FileVault login).

closed 2026-09-06 by dhalbert