adafruit/circuitpython #1140: CircuitPython 3.0.1 breaks built in NeoPixel on Feather M4 Express

ash0x1b, 2018-08-23:
After upgrading to CP 3.0.1, built in NeoPixel does't work anymore. It doesn't indicate the CP state with green/white and driving it with code has no effect.

Code below works in 3.0.0, doesn't work in 3.0.1

```python
import time
import board

import neopixel
led = neopixel.NeoPixel(board.NEOPIXEL, 1)
 
led.brightness = 0.3
 
while True:
    led[0] = (255, 0, 0)
    time.sleep(0.5)
    led[0] = (0, 255, 0)
    time.sleep(0.5)
    led[0] = (0, 0, 255)
    time.sleep(0.5)
```

jerryneedell, 2018-08-23:
I have also reproduced this issue on a feather_m4_express and on a metro_m4_express :-(

jerryneedell, 2018-08-23:
This is likely a result of PR #1132  -- I will experiment with the timing to try to resolve it.

deshipu, 2018-08-23:
Perhaps at some point we will need to add a parameter telling the neopixel code whether they are powered with 3.3V or 5V.

dhalbert, 2018-08-24:
@ash0x1b could you try this build, which includes #1141, and see if it works on your board? Thanks.

https://adafruit-circuit-python.s3.amazonaws.com/bin/feather_m4_express/adafruit-circuitpython-feather_m4_express-20180824-caa2328.uf2

dhalbert, 2018-08-24:
@ash0x1b Thanks! We'll do another release relatively soon.

dhalbert, 2018-08-28:
Fixed by #1141.

closed 2018-08-28 by dhalbert