adafruit/circuitpython #1147: nrf52 build failed with DEBUG=1 due to printf function

hathach, 2018-08-28:
As @dhalbert and @arturo182  suggested a couple of ways to resolve this. However, since this is in the main.c
https://github.com/adafruit/circuitpython/blob/master/main.c#L454

I better leave this to those who have more knowledge on the CP, since main.c changes will affect all port 

```
make V=1 BOARD=pca10056 DEBUG=1 all
GEN build-pca10056-s140/genhdr/mpversion.h
CC ../../py/modsys.c
CC ../../extmod/modwebrepl.c
CC ../../main.c
../../main.c: In function '__assert_func':
../../main.c:454:5: error: implicit declaration of function 'printf'; did you mean 'mp_printf'? [-Werror=implicit-function-declaration]
     printf("Assertion '%s' failed, at file %s:%d\n", expr, file, line);
     ^~~~~~
     mp_printf
cc1: all warnings being treated as errors
../../py/mkrules.mk:55: recipe for target 'build-pca10056-s140/main.o' failed
make: *** [build-pca10056-s140/main.o] Error 1
```

dhalbert, 2018-08-28:
I will fix this soon in a separate PR or as part of another.

dhalbert, 2018-10-03:
Fixed as part of #1158.

closed 2018-10-03 by dhalbert