adafruit/circuitpython #1155: CircuitPython standard library

notro, 2018-08-31:
I wanted to see if it was possible to use the CPython standard library on the M4 so I did a test and it looks promising. The unittest module which is probably one of the toughest wrt. demands on CircuitPython features did work with a small set of changes.

Here's the port of the os, time and unittest modules: https://github.com/notro/tmp_CircuitPython_stdlib

These are the CircuitPython changes: [stdlib.diff](https://gist.github.com/notro/e18fd3c43819e51f378ddc735c038c72)

The CP changes won't fit on the already overflowing boards, but for M4 it's a small change (1-2kB I think).

The next generation following the M4 will have even more RAM so I think this is the way to go, but the question is if this is the right time to do it.

ladyada, 2018-08-31:
neat! yes chips will only get faster and bigger :)

dhalbert, 2018-08-31:
@notro Do you know about https://github.com/micropython/micropython-lib and https://github.com/pfalcon/micropython-lib (forked due to some disagreements, but still close)? I'm not saying we should adopt those wholesale, but it may be interesting to see how some problems were solved.

notro, 2018-08-31:
@dhalbert Yes I have looked them. They try to be as small as possible, but my approach is to try and stay as close to CPython as possible since we are trying to be CPython compatible. The big benefit is that we get all the massively tested code for free, with just a small set of changes.

tannewt, 2018-08-31:
Cool! Awesome experiment.

dhalbert, 2018-08-31:
@notro Sounds good! Thanks for the background.

ladyada, 2018-09-01:
@tannewt @dhalbert do we want to have a circuitpython-lib? up to y'all :)

tannewt, 2018-09-04:
I would name it to match our current conventions. Perhaps it can live here: https://github.com/adafruit/Adafruit_CircuitPython_CPython

dhalbert, 2019-02-20:
Repointing some discussion that was in PR #1566 to here. Trying to summarize:

We decided to make weakly-linked `_` versions of a few modules that could be extended with Python versions derived from the CPython standard library. Right now, this is just `os` and `time`, though there could be more.

We have not enabled the "weak links" (module alias) mechanism in the M0 builds. There is not enough RAM or unused flash in the M0 builds to make using this practical, so it would be for the M4 and nRF52840 builds for now.

Some advantages and potential issues:
- The current subsetted modules can be made more complete by substituting fuller versions that are implemented in Python.
- In some builds, there is not enough room, as mentioned above.
- Providing such a mechanism means that there could be a significant split between how to use the smaller and larger builds, because people can write code that depends on the extended functionality in the larger versions. General-purpose library code should probably avoid using the extended modules, because then it won't run on smaller builds. That's also true of some libraries that might be highly useful but aren't extensions (such as `itertools`).
- The `_xxx` versions could possibly be confusing to novice users,though they won't see them except by doing `help('modules')`.
- There is significant spare space in flash in the nrf and M4 builds, which could be used to freeze in extended and new modules written in Python (like `itertools`). The advantage of freezing is that importing these modules wouldn't use up RAM. But we also want to leave room in flash for future capabilities that are CircuitPython-specific.

So there are some strategic decisions to be made here. We could provide the `_xxx` weak links mechanism, but not turn it on in the regular builds. If we do turn it on for all builds, then people might start to depend on it, and backing off from that choice later might be difficult.

More comments welcome. In the short term, we could leave this on experimentally by default in larger builds for 4.0.0 or turn it off as a default for now, to allow more time to see how our space requirements are going to change.

notro, 2019-02-20:
> Right now, this is just `os` and `time`, though there could be more.

These are also extended:
- [io](https://github.com/notro/tmp_CircuitPython_stdlib/blob/master/lib/io.py) (a couple of constants)
- [re](https://github.com/notro/tmp_CircuitPython_stdlib/blob/master/lib/re.py) (escape() and a constant)

```collections``` is also a likely candidate.

> * The `_xxx` versions could possibly be confusing to novice users,though they won't see them except by doing `help('modules')`.

It is a possibility to filter out the undersocre modules in [mp_help_print_modules()](http://elixir.tronnes.org/circuitpython/latest/ident/mp_help_print_modules) (maybe to the filtering in [mp_help_add_from_map()](http://elixir.tronnes.org/circuitpython/latest/ident/mp_help_add_from_map)).

tannewt, 2021-08-11:
Anyone want to keep this open? I don't think it's worth our time to try and port all of the stdlib. I'd rather have more targeted issues.

dhalbert, 2021-08-11:
The time to look at this kind of thing is when the bare-metal Pi port is done :)