adafruit/circuitpython #1160: Teensy builds are broken.

kdb424, 2018-09-04:
make: *** No rule to make target 'lib/mp-readline/builtin_input.c', needed by 'build/genhdr/qstr.i.last'.  Stop
I've been running micropython on my teensy, but couldnt' get HID support, so I hoped that CircuitPython would offer it. I tried to build it, but just failed. Is this no longer supported? Building for teensy 3.1

dhalbert, 2018-09-04:
Sorry, CircuitPython is not supported on Teensy. The builds that are supported are `atmel-samd` and `esp8266`, and `nrf` is under development.

closed 2018-09-04 by dhalbert

alexwhittemore, 2019-09-15:
@dhalbert just so I'm clear and not needlessly bashing my head against the wall, I'm getting the sense that the following is true:

- Circuitpython only supports atmel-samd, esp8266, and nrf, ergo there should be no expectation that any other board or port in this repo builds successfully
- By virtue of being a fork of MicroPython, a bunch of other, unsupported ports and boards exist in this repo
- No undue effort is taken to make sure unsupported boards or ports are purged
- In sum total, if anything not `atmel-samd`, `esp8266`, or `nrf` happens to build successfully, that is no indication that such will remain the case. And if it doesn't build successfully, I should go find that particular board/port in the upstream MicroPython repo, and use that.

Is that all correct? In other words, "not supported" boards are "an artifact of micropython heritage" rather than "we don't have time to help with those, but it might work"?

dhalbert, 2019-09-15:
You have it right. The supported ports now are `atmel-samd`, `nrf`, and `stm32f4` (in progress, and not the same as MicroPython's `stm32`). We also use the `unix` port for testing, but it does not try to emulate CircuitPython for regular use.

We don't purge other ports because i makes merging from upstream more difficult.

dhalbert, 2019-09-15:
BTW, support for `esp8266` was dropped as of 4.x.

alexwhittemore, 2019-09-15:
> and stm32f4 (in progress, and not the same as MicroPython's stm32)

That answers many many questions. Thanks!