adafruit/circuitpython #1162: NRF52840 Openthread Support

CraigZych, 2018-09-04:
Are there any plans to support OpenThread on the NRF52840 chip?
There are 2 different Mesh networking technologies natively built for this chip.

Both the nrf52832 and nrf52840 support Bluetooth5 Mesh but only the 840 supports thread. 

Personally thread support would be my first choice and I wish I knew where to begin exposing the api to upython. I've played around with OpenThread on the nrf52840 development boards and was very impressed. 

https://www.nordicsemi.com/eng/Products/nRF52840
https://openthread.io/
https://www.nordicsemi.com/eng/Products/nRF5-SDK-for-Thread

urish, 2018-12-11:
@tannewt I see you added support for Particle Mesh boards 👍, any news regarding Thread support?

tannewt, 2018-12-11:
Nope, I just added the board definitions to lay the foundation for someone else.

urish, 2018-12-11:
Got it, so still no concrete plans for Thread mesh at this time, correct?

tannewt, 2018-12-11:
Correct. We're currently focused on BLE and getting it in a stable release.

endico, 2021-01-03:
Here's example code for OpenThread on nRF52840. 
https://github.com/openthread/openthread/tree/master/examples/platforms/nrf528xx/nrf52840

The nRF5 SDK for Thread and Zigbee includes:

    * a pre-built OpenThread stack for the Nordic nRF52840 and nRF52811 SoCs,
    * support for hardware-accelerated cryptographic operations using ARM® CryptoCell-310,
    * software modules inherited from the nRF5 SDK e.g. peripheral drivers, NFC libraries, Bluetooth Low Energy libraries etc.
    * etc.