adafruit/circuitpython #1163: nrf: allow multiple I2C and SPI instances

dhalbert, 2018-09-04:
On nrf, TWIM0 and SPIM0, and TWIM1 and SPIM1 share the same hardware (kind of like SERCOMs on atmel). Right `now common-hal/busio/{I2C,SPI}.c` only support a a single instance.

There is an SPIM2 peripheral that is not shared with TWI.

Allow multiple SPI and I2C instances to be created.

On nRF52840, there is a third peripheral, SPIM3, which can run at a higher clock speed (32 MHz). One consideration is how to specify this interface. The SPI constructor currently does not allow specifying the clock speed. We could shuffle the SPI peripherals under the covers if someone asks for 32 MHz, assuming they were quiescent. Or we could add a new parameter to the `busio.SPI` constructor to specify the frequency at instantiation. `busio.I2C` already allows this.

dhalbert, 2018-09-04:
There's a discussion between @dhalbert and @arturo182 in discord 2018-09-04 starting 10:02 ET, pasted here:

**Dan Halbert**
@arturo182 also I will open issues to allow multiple instances for I2C and SPI. Right now only one instance is provided, assuming I'm reading the code correctly. Are there other periphs that provide only one instance but could provide more?

**arturo182**
I2C and SPI share same HW module for SPI1/I2C1 and SPI2/I2C2 and on nRF52840 there is also a unique SPI3 block, in the drivers the the i2c and spi instances are hardcoded so you can't use 2 SPIs at the same time for example
i think spi should prioritize SPI3 as first one because that's the only one that can do 32MHz
don't think any other modules share hw blocks
nrf52832 has 1 UART and nrf52840 has 2 uarts
but those are their own thing
ah and QSPI is a separate block too
so taht one i guess we can only use for exrternal flash
also we should have a issue for rewriting pwm in nrfx, it's a bit messy right now

**Dan Halbert**
Got it, I see the identical base addresses in 4.2.4 in the datasheet. so it's sort of like atmel SERCOMx except no UART. It would be nice if it were described as SPITWI1, etc. :/
we could choose SPI based on specified freq in the constructor
so you don't have to instantiate the high-speed one first to get it

**arturo182**
but the constructor doesn't take a speed parameter, it's in configure

**Dan Halbert**
oh blah, maybe it should, since having to know the order is going to cause support problems

**arturo182**
yeah, alternatively we could default to SPI1 and switch to 3 if the user calls configure with higher speed than supported by SPI1
since we don't use DMA then we know no operation is ongoing, so should be safe

**Dan Halbert**
yes, we could switch under the covers, since there's a full crossbar for the pins.

**arturo182**
yes, nrf is nice like that :smile:

**Dan Halbert**
i thought all periphs use easydma all the time

**arturo182**
yes they do but we use blocking operations

**Dan Halbert**
eventually ... we may not do that. this is worth discussing. unfortuately i can't attend today's audio mtg

**arturo182**
yes, but i think the dynamic switching is fine for now, we can rethink it later when we go for DMA

**Dan Halbert**
anyway I will open issues for to-do's and discussion

**arturo182**
but then we need a good async api

**Dan Halbert**
that is the $64k api

**arturo182**
sounds good, you can quote me in the issues :wink:

arturo182, 2018-09-04:
Really well put Dan!

dhalbert, 2018-09-20:
TWIM0 and TWIM1 are the same peripherals as SPIM0 and SPIM1. The IRQ handlers for these peripherals are set up at compile time, using some `#define` logic. and renaming. So out of the box TWIM0/SPIM0 and TWIM1/SPIM1 cannot be shared between `common-hal/busio/I2C.c` and `SPI.c`.

We could write an interrupt handler that checks whether it's being used for SPI or I2C,  nrfx uses weak references for the interrupt handlers, so it's possible to override them without modifying nrf. But for now two I2C's and 1-2 SPI's (1 in '832, 2 in '840) are good enough.

closed 2018-09-21 by tannewt

muhammadawais92, 2025-12-27:
### Issue
I’m using **CircuitPython on nRF52840** with **RFM9x (LoRa)** and an **SD card** sharing the same SPI bus.

My device behaves like a **Class C gateway**:
- LoRa is in continuous receive mode (`receive(timeout=0.1)`)
- Incoming sensor data is queued and logged to an SD card

SD writes frequently fail with:
OSError: [Errno 5] Input/output error

This happens even when:
- Logs are queued in RAM
- SD writes are retried later
- `spi.try_lock()` / `spi.unlock()` are used

It appears that the LoRa radio accesses SPI so frequently that the SD card never gets a clean window.

---

### Question
What is the **recommended approach in CircuitPython** to safely share SPI between:
- A continuously listening **RFM9x**
- An **SD card** used for logging?

Specifically:
- Should LoRa RX be temporarily stopped while writing to SD?
- Is batching SD writes required?
- Is there a known SPI access pattern or timing strategy for this case?

---

### Context
This project is a LoRa controller/gateway that must:
- Always listen for packets (Class C–like behavior)
- Reliably log data to an SD card

Any guidance or best practices would be appreciated.

Below is my code:

```python
import time
import board
import busio
import digitalio
import adafruit_rfm9x
import adafruit_ssd1306
import neopixel
import json
import storage
import adafruit_sdcard
import os
from adafruit_pcf8523.pcf8523 import PCF8523
from collections import deque

# ----------------------------
# LoRa setup
# ----------------------------
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.D11)
reset = digitalio.DigitalInOut(board.D12)
rfm9x = adafruit_rfm9x.RFM9x(spi, cs, reset, 433.0)

# ----------------------------
# OLED setup
# ----------------------------
i2c = busio.I2C(board.SCL, board.SDA)
oled = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c)

# ----------------------------
[... 162 lines cut ...]
                    queue_log()  # log regular sensor data
                elif data["type"]=="ACK":
                    if data["period"] == pending_period:
                        current_period = pending_period
                        waiting_for_ack = False
                        temp_message = "ACK received"
                        temp_message_end = time.monotonic() + 3
                        print("ACK received")
                        queue_log(ack=current_period)
            except Exception as e:
                print("Decode error:", e)

        # ---- Flush queued logs to SD ----
        flush_log_queue()

        # ---- Update OLED ----
        update_oled()

except KeyboardInterrupt:
    oled.fill(0)
    oled.show()
    pixels.fill((0,0,0))
    print("Program stopped, OLED cleared.")

```

muhammadawais92, 2025-12-27:
@dhalbert can you please have a look at my particular issue? Thank you in anticipation.

dhalbert, 2025-12-27:
@muhammadawais92 Please start a new issue. The original issue was fixed long ago. If you can try a second SPI bus (which is available) that will be a good diagnostic.

muhammadawais92, 2025-12-27:
Thank you @dhalbert i will try it. If the issue presists i will start a new issue.

muhammadawais92, 2025-12-27:
@dhalbert I have created a new issue.
https://github.com/adafruit/circuitpython/issues/10766

PR #1203 (2018-09-21): nrf: multiple i2c and spi instances; improve nrfx_config.h
Fixes #1163.

- Allows allocation of TWIM0 and TWIM1 on nRF52832.
- Allows allocation of SPIM2 on nRF52832.
- Allows allocation of SPIM3 and SPIM2 on nRF52840. SPIM3 is allocated first because it's higher speed (32 MHz).
- We could conceivably allocate SPIM0 and SPIM1, but we have to fix the IRQ handlers, which are defined at compiled time, depending on whether SPIM0 or TWIM0 (or SPIM1 or TWIM1) is used.. So I didn't because two SPIs is probably enough. See https://github.com/adafruit/circuitpython/issues/1163#issuecomment-423318976 fo [... cut]