adafruit/circuitpython #1170: Support scrolling by changing the portion of the framebuffer read by the display

tannewt, 2018-09-07:
This is particularly helpful for scrolling text output so that the new line only needs to be transferred.

The gameboy uses this technique. Its explained here: https://youtu.be/HyzD8pNlpwI?t=31m43s

tannewt, 2018-09-10:
Really? The PTLAR only affects new pixels? I haven't tested it yet though.

deshipu, 2018-09-11:
Yes, the datasheet is not clear about this, but it controls how the pixels are written by the memory write commands, not how they are sent to the display. At least that's what I figured out with my experiments, I might remember it wrong, it was a year ago.

deshipu, 2018-09-11:
Actually, that wasn't PTLAR. PTLAR is for only using a part of the display (for example, when the rest is covered by the bezel).