adafruit/circuitpython #1175: Stack size problems

arturo182, 2018-09-08:
Like I mentioned on Discord, after updating my code from somewhere around July to current master, I ran into a problem where a import was failing with "ImportError: cannot import name Xyz". After a lot of bisecting I managed to identify that this problem started with #1057 

While I didn't dive too deep into what exactly is happening, I managed to fix the issue by increasing the `CIRCUITPY_DEFAULT_STACK_SIZE` value from 4K to 8K.

dhalbert, 2018-09-08:
In 3.0 and SAMD21, the default stack is 12k, including a 1k guard region. So yeah, that just sounds too low -- we should increase it. @tannewt?

I thought Scott and I discussed this in a PR but I can't find it.

dhalbert, 2018-09-08:
But it's still odd - should give a more straightforward error.

tannewt, 2018-09-11:
@dhalbert Increasing works for me. Its possible this is the error when the parse stack runs out of space since it's not actively running python code.

notro, 2018-09-11:
I need 8k to run the [time_os](https://github.com/notro/tmp_CircuitPython_stdlib/blob/master/lib/test/test_os.py) and [time_time](https://github.com/notro/tmp_CircuitPython_stdlib/blob/master/lib/test/test_time.py) unittests.
For the tests that test unittest itself 8k isn't enough so I use 12k.

tannewt, 2019-02-01:
I haven't run into stack sizing issues. We now check to see if the stack overran the heap and enter safe mode if that is the case. (Throwing an exception is impossible because the Python code itself could have been overwritten.)

closed 2019-02-01 by tannewt