adafruit/circuitpython #1176: MacOS: "unidentified keyboard" from USB HID

dhalbert, 2018-09-09:
When CircuitPython starts on a Mac and presents an HID keyboard device, MacOS brings up a the Keyboard Setup Assistant saying the keyboard can't be identified. After a few times, it seems to stop asking. But it would be nice to prevent this from happening at all. Investigate whether we can set something in the keyboard descriptor that prevents this.

tannewt, 2018-09-10:
We could also disable it by default once we have dynamic descriptors hooked up. Then it'd be clear when it happens.

dhalbert, 2021-09-09:
I asked how keyboards get identified in the Apple Developer Forums (https://developer.apple.com/forums/thread/687750) and did not get any useful answers.

dhalbert, 2025-01-31:
I think this is unsolvable and will close this for now.