adafruit/circuitpython #1179: How are class function objects created?

notro, 2018-09-09:
unittest uses decorators to add properties to functions about skipping and if it's expected to fail.
[MicroPython doesn't support properties on functions](http://docs.micropython.org/en/latest/unix/genrst/core_language.html#user-defined-attributes-for-functions-are-not-supported), so the work around is to use a dictionary with the function object as the key.

However I'm getting different function objects for the same function depending on which scope I'm in:
```
Adafruit CircuitPython patchbase-7-g216f0f952-dirty on 2018-09-09; Adafruit Metro M4 Express with samd51j19
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class A:
===     def a(self):
===         pass
===     print(a)
===
<function a at 0x20002990>
>>>
>>> A.a
<function a at 0x2002d8b0>
>>> A.a
<function a at 0x2002d8b0>
>>>

```
Does anyone know what's going on here?

I've tried to turn on some debug output, but I haven't been able to find where that ```A.a``` function object is created:
```
=== class A:
===     def a(self):
===         pass
===     print(a)
===
assign byte code: code=20002a40 len=27 flags=0
assign byte code: code=20002b20 len=44 flags=0
assign byte code: code=20002b60 len=17 flags=0
make_function_from_raw_code 20002990
mp_obj_new_fun_bc: 20002960
make_function_from_raw_code 2002d970
mp_obj_new_fun_bc: 200028f0
make_function_from_raw_code 20002a90
mp_obj_new_fun_bc: 20002990
<function a at 0x20002990>
>>>
>>> A.a
assign byte code: code=20002a70 len=28 flags=0
make_function_from_raw_code 20002a00
mp_obj_new_fun_bc: 200029d0
<function a at 0x2002d8b0>
>>>

```

deshipu, 2018-09-09:
Well, the "." operator is magical for functions. If called on a class, it will return a wrap the function in an instance of a class method, so that it can remember from which class you took it. If called on an instance, it will wrap the function in an instance bound method, so that it can keep a reference to the "self".

tannewt, 2018-09-10:
@notro The different address could be due to the long-living of types. Maybe track that too?

tannewt, 2018-09-10:
I only planned for it to long live the reference accessible through the type. You've definitely hit a corner case where it doesn't work. https://github.com/adafruit/circuitpython/blob/master/py/objtype.c#L1056

I don't understand what you are trying to achieve enough to offer a fix.

notro, 2018-09-16:
I've now been through all the tests that test unittest itself (except mock).
```unittest.expectedFailure``` is adding a property on the function and this is working using the function name as dictionary key.
The other use case is functions marked with ```unittest.skip```. Skip itself raises an exception so that is working. But if all functions in a class is marked skipped, unittest also skips calling ```setUp```, so this functionality is lost. This is no big deal, since the only real loss it that the test takes a bit longer to run. 
Thanks for the help.

closed 2018-09-16 by notro