adafruit/circuitpython #1209: SPI peripherals not cleared on soft reboot

jerryneedell, 2018-09-23:
Just stumbled across this with current master on a nrf52840 Dongle  - not sure if it occurs on other boards -- will test later. Since there has been recent work on pin usage I thought I post this now then add more as time for testing allows.

After 2 successful runs mounting  the SPI sdcard then doing a soft reboot (control-D) I get the following error on the third and subsequent try:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-21; PCA10059 nRF52840 Dongle with nRF52840
>>> import sdmount_lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount_lib.py", line 8, in <module>
ValueError: All SPI peripherals are in use
>>> 
```

the code being run is:
```

import adafruit_sdcard
import busio
import digitalio
import board
import storage
import sys
# Connect to the card and mount the filesystem.
spi = busio.SPI(board.P0_17, board.P0_24, board.P0_20)
cs = digitalio.DigitalInOut(board.P1_10)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
sys.path.append("/sd")
sys.path.append("/sd/lib")

```

jerryneedell, 2018-09-23:
NOTE: a hard reset clears it all up -- and the mount succeeds.

jerryneedell, 2018-09-24:
I have reproduced this issue on an nrf52840 pca10056.

```>>> import sdmount_lib
>>>
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-21; PCA10056 nRF52840-DK with nRF52840
>>>
>>> import sdmount_lib
>>>
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-21; PCA10056 nRF52840-DK with nRF52840
>>>
>>> import sdmount_lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount_lib.py", line 8, in <module>
ValueError: All SPI peripherals are in use
>>> 
```

I also verified that this problem is new. It did not occur on the previous build that was on the pca10056 from 9/14/2018.

jerryneedell, 2018-09-24:
This issue does NOT  occur on my metro_m4_express.

dhalbert, 2018-09-24:
Thanks for finding this. Not surprised based on my recent additions and probably easily fixed.

jerryneedell, 2018-09-24:
on the feather_nrf52832 this issue occurs after the first successful run (on the Dongle it ran twice then failed)  Not this also worked normally on a build from 9/21/2018

```
ress any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-24; Bluefruit nRF52 Feather with nRF52832
>>> 
>>> 
>>> 
>>> import sdmount_lib
>>> 
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-24; Bluefruit nRF52 Feather with nRF52832
>>> 
>>> import sdmount_lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount_lib.py", line 8, in <module>
ValueError: All SPI peripherals are in use
>>> 
```

jerryneedell, 2018-09-24:
This appears to have been introduced PR #1203.

jerryneedell, 2018-09-25:
appears to be fixed by #1215

closed 2018-10-01 by dhalbert