adafruit/circuitpython #1210: Delayed access to REPL session under Linux

jerryneedell, 2018-09-24:
After upgrading my Linux system to UBUNTU 18.04 I noticed that there was a long (10s of seconds) delay before I was able to connect a terminal session to /dev/ttyACM0. The CIRCUITPYTHON drive would mount immediately but attemps to connect with
```
screen /dev/ttyACM0 115200
```
would fail unless I waited 10 - 20 seconds to attempt it. After the delay, it worked fine.

As @dhalbert suggestion I removed the "modem manamger"
```
sudo apt-get remove modemmanager
```

and the problem is resolved. I tied a metro_m4_express and an nrf52840 Dongle and both responded to the screen connection immediately.

dhalbert, 2018-09-25:
I tried this on an Ubuntu 18.04 machine with modem-manager installed, and it had the delay as reported. This is strange because there's a udev rule entry:
```
ATTRS{idVendor}=="239a", ENV{ID_MM_DEVICE_IGNORE}="1"
```
I confirmed that the device has this rule active. It appears modem-manager is no longer doing the right thing and ignoring the device as it's supposed to.

I'll close this issue after documenting disabling and/or uninstalling modem-manager in one or more Learn Guides.

dhalbert, 2018-10-01:
Documented how to remove modemmanager in https://learn.adafruit.com/welcome-to-circuitpython/kattni-connecting-to-the-serial-console

closed 2018-10-01 by dhalbert